meshtastic/Meshtastic-Android #14: In app pairing doesn't work without location services on.

Dafeman, 2020-04-12:
If location services are turned off then no devices are shown and you cannot pair.  In older app versions I think there was a warning about needing both turned on but not in recent releases. If possible it would be good not to require location services for boards that are equipped with GPS already.

geeksville, 2020-04-12:
Alas, in Android (at least for all recent releases - I don't remember
without checking the devdocs when they added this restriction):  Because
being able to see bluetooth devices allows android apps (especially
'sneaky' apps that might be trying to secretly gleam information about
where the users is) conveys information to the app developer that might
reveal the user's location.  From the docs:

> The other permission that you must declare is ACCESS_FINE_LOCATION
<https://developer.android.com/reference/android/Manifest.permission#ACCESS_FINE_LOCATION>.
Your app needs this permission because a Bluetooth scan can be used to
gather information about the location of the user. This information may
come from the user's own devices, as well as Bluetooth beacons in use at
locations such as shops and transit facilities.

If the warning about requiring that location permission is not in there we
should add it back in.  Because if we don't get it Android doesn't let us
access the bluetooth API.

geeksville, 2020-04-13:
Also your comment made me realize I made a big mistake in my latest
release.  I suspect you will find it amusing.  I'm rushing out a new build
to fix this:

https://github.com/meshtastic/Meshtastic-Android/commit/51a12450689700f680bc03903a68075e980070c7#

Dafeman, 2020-04-13:
Ah right, seems overboard to require location be turned on just so the user is now aware that their location may be known when using BT! This may be useful for newer devices as it doesn't require location permissions. https://developer.android.com/guide/topics/connectivity/companion-device-pairing 
Disclaimer - I have no programming experience, just google search experience!

geeksville, 2020-04-13:
oh!  that's great!  Thanks for the link - I hadn't heard they added that.  I'll repurpose this bug:

> On devices running Android 8.0 (API level 26) and higher, companion device pairing performs a Bluetooth or Wi-Fi scan of nearby devices on behalf of your app without requiring the ACCESS_FINE_LOCATION or BLUETOOTH_ADMIN permission. This helps minimize the number of permissions needed by your app and is more privacy-friendly for users. After the device is paired, the device can leverage the REQUEST_COMPANION_RUN_IN_BACKGROUND and REQUEST_COMPANION_USE_DATA_IN_BACKGROUND permissions to start the app from the background. You might use this method of pairing devices for initial configuration of a BLE-capable smart watch, for example.

Companion device pairing doesn't create the connection to chosen devices. You use Bluetooth or Wi-Fi connectivity APIs to establish connections. Companion device pairing also isn't intended for continuous scanning. It is a means for implementing custom protocols over Bluetooth, BLE, and Wi-Fi that are required for the companion device to function.

The requested permissions are granted when the user associates your app by selecting a device from the list. These permissions are revoked only if you call disassociate() or if the app is uninstalled. An app is responsible for clearing its own associations if the user no longer needs them, such as when they log out or remove bonded devices.

Dafeman, 2020-04-13:
Awesome, love your work!

closed 2022-01-10 by mc-hamster