meshtastic/Meshtastic-Android #19: Support off-line maps

geeksville, 2020-04-22:


geeksville, 2020-04-22:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/offline-maps-on-meshtastic/137/3

aHVzY2g, 2020-06-11:
I would suggest the usage of OsmAnd. The only job to be done would be to integrate the "aidl interface", of Osmand. https://github.com/osmandapp/Osmand/blob/master/OsmAnd/src/net/osmand/aidl/IOsmAndAidlInterface.aidl#L445

geeksville, 2020-06-12:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/offline-maps-on-meshtastic/137/21

aHVzY2g, 2020-06-26:
@geeksville did you already had a thought, which "OfflineMap integration" would fit the scope of the project best?

geeksville, 2020-06-27:
@aHVzY2g Mapbox as a really nice and easy looking API to turn on off-line maps.  I just didn't call it ;-).  So I'd say the "low hanging fruit" solution is for someone to just use the mapbox version.  

Porting to some other API would be fine if someone wants to do that and it provides the same level of polish.  But the docs for mapbox are super good and it was super easy to for me add maps with their api.  the last time I looked at OsmAndroid it was much more rough - but that was for a project like 6 yrs ago (andropilot and droneshare) so it has probably gotten better.

jetatomic, 2020-07-11:
Please consider non-urban areas when looking at map examples. Many maps which look very cool in cities end up being completely blank in areas with no roads. OpenStreetMap, which is the source OsmAnd primarily draws from, only has 1 layer (of 5) which shows useful details when there are no roads. If OSM sources are selected, please consider using the _Cycle Map_ layer if no other robust data sources are layered over the basemap. Otherwise, people hiking might find this is all the map shows, which is the 4km surrounding my home:

[image]

[Example in the Meshtastic forum](https://meshtastic.discourse.group/t/offline-maps-on-meshtastic/137/24?u=jetatomic).

chron0, 2020-08-10:
Offline Maps would seem to be a key feature for the idea of meshtastic. OsmAnd or the principle of OsmAnd could be a viable alternative, as well as having in app capability to download and store maps when online and use them when offline. QGroundControl has a similar feature like this, where you can use basically any source, vector or sat maps, create a box and it will download all resolutions you specify (and have available space).

geeksville, 2020-08-10:
@chron0 It might be easier to initially just turn on offline maps in our current mapbox client.  If you want to work on this, here's a plausible pointer: https://docs.mapbox.com/help/troubleshooting/mobile-offline/

chron0, 2020-08-10:
I'll have a look but, although the mapbox stack is potent/competent, I never liked their closed box model. A couple of years back I was saddling this horse the other way around: https://apollo.open-resource.org/lab:dspace - but in the end we failed at the challenge of the distributed data layer. meshtastic could proof to provide this kind of basis the other way around where we could map communication/data in a decentralized (mesh) fashion as "overlays" onto an available basemap.

DingoOz, 2020-09-07:
Additional link to MapBox API calls for offline maps: https://docs.mapbox.com/android/maps/overview/offline/

cjgemm, 2021-01-22:
Tried this, but the downloaded maps do not work in the Meshtastic app.  Osmand is the nicest non ATAC mapping app I have found.

geeksville, 2021-03-10:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/can-the-app-use-other-maps/246/6

joshbowyer, 2022-01-24:
As a "power user" of Osmand, it is definitely far ahead of ATAK. And as for data sparsity, enhanced mapping data (tiles, POIs, routing, etc) can be downloaded from opensupermaps.com and placed in Osmand's working directory. With this dataset Osmand becomes the best offline mapping software available IMO.

Mhowser, 2022-01-30:
@geeksville are these docs up to your standards?

https://github.com/osmdroid/osmdroid/wiki#kotlin
It would be awesome if this could be done for #6's sake as well.

geeksville, 2022-01-30:
Sure, if someone wants to send in a PR to do this then I think that would be great!

Mhowser, 2022-01-31:
I am trying to commission a person on Fiverr to do this.

joshbowyer, 2022-02-22:
I dont know how Fiverr works but is it something where users like me could donate to a bounty pool? I personally know some other people aside from just myself who would be willing to pay for this to be made

joshbowyer, 2022-03-19:
Have you had any luck?

Mhowser, 2022-03-19:
No, all devs that I contacted there basically told me its too complicated. I could ask more devs but most of them are out of my budget.

joshbowyer, 2022-03-19:
May I ask what the budget is? Because Im serious about wanting to pitch in with a few people, we could discuss it via email too if youd prefer

Mhowser, 2022-03-20:
My budget was around 200 at the time, but since then I had to spend over half my savings due to mandatory life style changes. So I'm afraid I won't be able to contribute for quite a while.

joshbowyer, 2022-03-20:
I am looking to put forward a few thousand.

Email me at mke_josh@protonmail.com and we can discuss more

Mhowser, 2022-03-20:
@andrekir I highly suggest you take advantage of this opportunity.

PWRxPSYCHO, 2022-03-20:
I ended up making a prototype of offline-maps using MapBox. It was merged in with PR #391 and was linked to Issue #359 .

It is a "developer only feature" right now and is very limited in its functionality. 

The main issue is MapBox's API limits for the free tier, as well as the Terms of Service which prevents users from downloading the tilesets and keeping them stored permanently. You must fetch the tilesets before going into a region with no cell coverage. It essentially "pre-loads" the tilesets in cache and allows you to view the specified region when you need it. (As long as the application is still open and has not be closed)

Ideally, a transition away from MapBox would be the best solution to this problem. But that has its own set of issues, namely performance on older devices, actually setting it up in the application, and other unknowns.

joshbowyer, 2022-03-20:
Please look at Osmand/OSM. The issue youre running into is the same issue PureMaps has been having; due to limited API calls theyve had to remove nearly every tile server they had.

PWRxPSYCHO, 2022-03-20:
I could probably take this on, but it won't be a quick turnaround as I would be doing this in my free time lol. I would also need to talk with the other devs on the project and see if this would be a viable solution before I spend too much time working on something like this.

chron0, 2022-03-20:
Something like Meshtastic will only unfold its true potential when it can work without any online dependency. So integrated offline-maps coming from OSM and/or re-using OSMAnd map downloads and serving it locally in app, or providing it's own tileserver that the client app can use as a drop-in replacement would be awesome.

prurigro, 2022-08-05:
Would interfacing with the osmand API be a valid solution?

This is an example app, and the readme has the list of commands with their arguments and explanations: https://github.com/osmandapp/osmand-api-demo/tree/master/OsmAnd-api-sample

It seems like `addMapMarker()`, `updateMapMarker()` and `removeMapMarker()` could be used to draw, update and remove the locations of other nodes, which would be great for initial functionality. It also seems like there are enough features available to extend the functionality quite a bit down the road.

If osmand is available as an option for mapping, it might make sense to replace the mapbox functionality with a list of nodes and their GPS coordinates? The app would be fully functional without the internet, and the coordinates would be easy to reference and punch into other GPS devices.

closed 2022-10-08 by andrekir