meshtastic/Meshtastic-Android #20: Unable to pair bluetooth using Android phones

android606, 2020-04-25:
Using both a (reasonably new) Motorola Moto G7 with Android 9, and a (super old) HTC One M7 with Android 4.4.

I tap "pair" on the phone, then sometimes get no response at all, other times get a prompt on the phone for a passkey, but get no key displayed on the device or serial port.

Also noticed the "signal xx seconds ago" display often resets to 0 at the instant you tap "pair".

geeksville, 2020-04-25:
@android606 hmm - I don't see a report yet in my console.  Would you mind launching the app one more time (the google crashlytics actually doesn't send the crash report until the next time the user launches the app)

android606, 2020-04-26:
I've tried clicking "Report Bug", then restarted the app, a couple times. I'm not 100% sure it's successfully sending a report...

android606, 2020-04-27:
Come to think of it, this is a problem between the phone and device, not the app. It should probably be an issue for Meshtastic-device.

geeksville, 2020-04-28:
@android606 though if history is any guide, almost certanly whatever the bug is probably in the phone app side ;-).  The device side seems fairly solid.

re: reports
Yeah, I haven't seen a report come through for a G7.  A few questions to help figure out why reports aren't getting sent:
a) is your phone rooted?
b) do you use any sort of network wide 'ad blocker' (pihole etc...) at your site?
c) do you have any sort of anti-adware filter on the phone itself? 
(Most of thost analytics blockers also block connections to crashlytics)

android606, 2020-04-28:
Haha, gotcha! :-)

a) My Moto G7 was _not_ rooted when I first encountered this problem, but I rooted it yesterday, and the problem is still occurring. 
My HTC One is not rooted.
b) No, but I'll try it again with the phone on LTE instead of wifi, just to eliminate the possibility of a router issue.
c) No, both phones were bone stock and freshly factory-resetted.

Is there any way to manually collect the same information Crashlytics provides? Maybe capture a logcat?

android606, 2020-04-28:
Just tried again, pairing and then submitting a report. Behavior on this end seems the same as previous attempts.

After attempting to pair 3-4 times, a toast appears that says "Could not pair because of incorrect pin or passkey", even though the phone never prompted me for a pin, and one never appeared on the device.

geeksville, 2020-04-28:
I'll reply at length later when at keyboard but.  Could you try going to
Android settings / Bluetooth and choose pair new device and see if you see
your radio there?  And they if you click on it there it then prompts for
the pin shown on the radio screen?

(Sent from a phone - please ignore typos)

android606, 2020-04-29:
That's exactly the problem- sorry, maybe I wasn't clear about that. I'm trying to pair from Android Settings/Bluetooth. The device(s) show there, but when I attempt to pair, I get "Could not pair because of incorrect pin or passkey". The phone never prompts me for a pin, and the device never shows me a pin on the screen. 

Often, but not always, the update timer on the device OLED resets to 0 seconds right at the moment I attempt to pair.

geeksville, 2020-04-29:
oh thats interesting!  yes.  That implies something device side and the android log wouldn't be useful in that case.  Would you mind pasting a log from the serial console on the device when this happens?  Also, what version # for the device code?