meshtastic/Meshtastic-Android #21: position on mapview does not update

claesg, 2020-04-26:
I took a longer walk and noticed that the position on the mapview does not update. It shows the position where the app was started. Restarting the app will update the position.
At the time of testing there was only the own device in the net. Other devices were out of range.
I did push 'report bug' button in the app.
Running app 0.5.7 and device 0.4.3 (TTGO T-BEAM1.0)
Phone is a Oneplus 5T

geeksville, 2020-04-26:
thanks for the report.  I'll look into this later this week.  Also, I just fixed an autobug for a Oneplus 5T (#23) that you might not have noticed but it will be in the next release. (Probably unrelated to to this but I haven't looked yet)

geeksville, 2020-05-24:
fixed a long time ago

closed 2020-05-24 by geeksville