meshtastic/Meshtastic-Android #24: Misc app improvement ideas

geeksville, 2020-05-03:
This post includes both device and app feature ideas (but mostly for the app side).

I'm making an issue here and we should think about this/start work on them after the 1.0 release is out.

https://meshtastic.discourse.group/t/various-function-suggestions/161/7

geeksville, 2020-05-03:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/various-function-suggestions/161/7

Xanuman, 2020-05-24:
This project is perfect for use as a gps tracker. You can track the whereabouts of relatives (children and the elderly) or pets.

In particular, I plan to use Meshtastic to control the flock of sheep. In this regard, I had a suggestion for improving the Android application:

The ability to configure geofences in the application. For example, we indicate a network node and assign it a geofence. When a node leaves the geofence, the application gives us an alarm.

JamesM92, 2020-10-15:
I believe its been mentioned on the forum but not here. But the ability to send messages to an individual and not the entire group would be useful.

andrekir, 2022-02-22:
many features listed were added since and this feature request cycle is done. closing.

closed 2022-02-22 by andrekir