meshtastic/Meshtastic-Android #26: autobug - 'crash' on switching between devices

geeksville, 2020-05-04:
from @erikthev-king

> wow! upgraded to 0.6.0 an hour ago and had a chance to play around with 2 tbeams. Perfect brew! Crashed once while swaping phone connection over from one tbeam to the other one but I guess it was my phone’s bluetooth messing up. Hope you got the crash report, Kevin.

[com.geeksville.mesh_issue_2a12c7f9d75db78e3f3f9648aff2177a_error_session_5EAFBD9D0204000177D962CEF307FD08_DNE_0_v2.log](https://github.com/meshtastic/Meshtastic-Android/files/4575505/com.geeksville.mesh_issue_2a12c7f9d75db78e3f3f9648aff2177a_error_session_5EAFBD9D0204000177D962CEF307FD08_DNE_0_v2.log)

geeksville, 2020-05-24:
fixed in 774ab65ece6f5c8b2c298d5626e4826d5cd123e3

closed 2020-05-24 by geeksville

eriktheV-king, 2020-05-24:
Great ! thumbs up ! Thanks !

commit 4efd25d0 (2020-05-24): Possible fix #26.  It seems some phones don't quite have the services
list populated at the time they say service discovery has completed.
Make the nasty 500ms delay a 1000ms delay instead