meshtastic/Meshtastic-Android #29: autobug analysis TBD

geeksville, 2020-05-21:
@vaderj sez:

> I have uninstalled and reinstalled the app multiple times (Android app v0.6.4) on my HTC 10 (Android 7.0, no SIM, Wifi only) I do not have this problem on my HTC U12+ I hit the Report Bug associated with device Meshtastic_db6c Once in the app and the app claims to be paired, nothing really happens and at no matter where I touch any of the screens, nothing really happens edit So from my HTC 10, the cloud icon in the top-right corner has a cross through it, though it is connected to wifi whic…

I'll pull the logs from google shortly and update this bug.

geeksville, 2020-05-21:
[com.geeksville.mesh_issue_a91178babb9fbe1de0c2c5d2f51c1cae_error_session_5EC194380252000104E79B6BFB2DEAC5_DNE_1_v2.log](https://github.com/meshtastic/Meshtastic-Android/files/4663828/com.geeksville.mesh_issue_a91178babb9fbe1de0c2c5d2f51c1cae_error_session_5EC194380252000104E79B6BFB2DEAC5_DNE_1_v2.log)

geeksville, 2020-05-21:
hmm - from that log it looks like the OStold the app about he device when it did the BLE scan, but the connect attempt never succeeded:

```
73 | Sun May 17 2020 13:04:08 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.ui.BTScanModel: onScanResult BTScanEntry(name=...:AC, addr=...:AC)
74 | Sun May 17 2020 13:05:08 GMT-0700 (Pacific Daylight Time) | com.geeksville.analytics.GoogleAnalytics: Analytics: end screen
75 | Sun May 17 2020 13:05:08 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.ui.BTScanModel: stopping scan
76 | Sun May 17 2020 13:05:08 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.MainActivity: Unbinding from mesh service!
77 | Sun May 17 2020 13:08:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.MainActivity: Binding to mesh service!
78 | Sun May 17 2020 13:08:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.MeshService$Companion: Trying to start service
79 | Sun May 17 2020 13:08:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.analytics.GoogleAnalytics: Analytics: start screen Settings
80 | Sun May 17 2020 13:08:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.ui.BTScanModel: BTScan component active
81 | Sun May 17 2020 13:08:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.ui.BTScanModel: starting scan
82 | Sun May 17 2020 13:08:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.MeshService: in connectionState=DISCONNECTED
83 | Sun May 17 2020 13:08:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.MeshService: in getOnline, count=0
84 | Sun May 17 2020 13:08:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.MainActivity: Ignoring failure to get myId, service is probably just uninited... We don't yet have our myNodeInfo
85 | Sun May 17 2020 13:08:38 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.MainActivity: connchange DISCONNECTED
```

I'll update this bug once I look at this code (today)

geeksville, 2020-05-21:
btw - can you do me a favor and try a test of rebooting your phone and see if it now happily connects to that radio?

vaderj, 2020-05-22:
Restarted HTC 10, uninstalled and reinstalled Meshtastic from the Play Store, removed all BT devices and tried again - I was prompted and able to enter the BT pin, but then I got the cloud with a cross (presumably, disconnected?) and nothing else.  I also sent more logs

vaderj, 2020-05-22:
So I uninstalled, reinstalled again (probably the 5th time now?) and for some reason, it decided to work this time.

I had to disconnect and reconnect from the other device with my other android to get it to update the new node.

Thanks!

vaderj, 2020-05-22:
At the point I ran the test, I think both phones were running .6.5 of the Android app, but both of my TTGO devices were on .6.3 of device firmware.

At that point I was able to connect with my HTC 10 and send a message from both devices and it propagated correctly, but since then I have not been able to get any messages to send (just from the app, and clicking the middle button on the T-Beams)

I just updated to .6.4 on the T-Beams and it appears both phones connect to their own TBeams, but I have not been able to get pass any messages to the android devices, even though I see the messages appear on both TBeam screens.

I have found when testing the app on Android, it helps to, from Android settings, force stop the app, clear (just) the app cache, and possibly reset the applicable TBeam before restarting the Meshtastic app.

Thanks and let me know what I can do to help!  I have a 3rd TBeam on the way from China and I am going to give one to a buddy who lives a mile away.

vaderj, 2020-05-22:
Additionally, I forgot to mention that the only messages that propagate to my other android devices do not have an accurate timestamp.  Once texts on my android have the correct timestamp, they don't seem to be received my the other android device - I will do more testing tonight / tomorrow to see if I can define that behavior better

geeksville, 2020-05-24:
Here's a really good log from @vaderj's HTC 10.  Much more useful than the previous one.  I'll look into this tomorrow morning, drinking now.

[com.geeksville.mesh_issue_a91178babb9fbe1de0c2c5d2f51c1cae_error_session_5EC81133009B00010923671574BB8DA0_DNE_0_v2.log](https://github.com/meshtastic/Meshtastic-Android/files/4672850/com.geeksville.mesh_issue_a91178babb9fbe1de0c2c5d2f51c1cae_error_session_5EC81133009B00010923671574BB8DA0_DNE_0_v2.log)

geeksville, 2020-05-24:
@feh123, what is your phone model name - so I can find the google autobug report?

feh123, 2020-05-24:
@geeksville a Sony Z1 model number C6903. Rather old now!

geeksville, 2020-05-24:
report from @feh123 phone...[com.geeksville.mesh_issue_a91178babb9fbe1de0c2c5d2f51c1cae_error_session_5ECA1BEB00BA00013753AA4B1EC99EF8_DNE_0_v2.log](https://github.com/meshtastic/Meshtastic-Android/files/4674137/com.geeksville.mesh_issue_a91178babb9fbe1de0c2c5d2f51c1cae_error_session_5ECA1BEB00BA00013753AA4B1EC99EF8_DNE_0_v2.log)

geeksville, 2020-05-24:
@feh123 the problem on your phone seems to be:

```
223 | Sun May 24 2020 00:15:04 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: new bluetooth connection state 2, status 0
224 | Sun May 24 2020 00:15:04 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: work connect is completed, resuming status=0, res=kotlin.Unit
225 | Sun May 24 2020 00:15:04 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.RadioInterfaceService: Connected to radio!
226 | Sun May 24 2020 00:15:04 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: Enqueuing work: reqMtu
227 | Sun May 24 2020 00:15:04 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth$BluetoothContinuation: Starting work: reqMtu
228 | Sun May 24 2020 00:15:12 GMT-0700 (Pacific Daylight Time) | com.geeksville.analytics.GoogleAnalytics: Analytics: end screen
229 | Sun May 24 2020 00:15:12 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.ui.BTScanModel: stopping scan
```

When we make the call to change the mtu (a bluetooth blocksize thing), your phone never completes that call at all - doesn't return either error or success.  By coincidence I noticed this in an anonymous report from some other phone.  I've implemented a possible fix in bug #27.

Can you try the build I release later today and reopen a bug if is still not working for you?  Also, when cycling between various devices on the settings page, please wait 20ish seconds before trying a different one (it will make it easier for me to read the logs if you press report bug).

geeksville, 2020-05-24:
@vaderj your bug also is definitely #27 

```
 48 | Fri May 22 2020 10:52:53 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: new bluetooth connection state 2, status 0
 49 | Fri May 22 2020 10:52:53 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: work reconnect is completed, resuming status=0, res=kotlin.Unit
 50 | Fri May 22 2020 10:52:53 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.RadioInterfaceService: Connected to radio!
 51 | Fri May 22 2020 10:52:53 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth: Enqueuing work: reqMtu
 52 | Fri May 22 2020 10:52:53 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.service.SafeBluetooth$BluetoothContinuation: Starting work: reqMtu
 53 | Fri May 22 2020 10:52:54 GMT-0700 (Pacific Daylight Time) | com.geeksville.analytics.GoogleAnalytics: Analytics: end screen
 54 | Fri May 22 2020 10:52:54 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.ui.BTScanModel: stopping scan
 55 | Fri May 22 2020 10:52:54 GMT-0700 (Pacific Daylight Time) | com.geeksville.analytics.GoogleAnalytics: Analytics: start screen Messages
 56 | Fri May 22 2020 10:52:54 GMT-0700 (Pacific Daylight Time) | screen_view
 57 | Fri May 22 2020 10:53:03 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.ui.BTScanModel: BTScan component active
 58 | Fri May 22 2020 10:53:03 GMT-0700 (Pacific Daylight Time) | com.geeksville.mesh.ui.BTScanModel: starting scan
 59 | Fri May 22 2020 10:53:03 GMT-0700 (Pacific Daylight Time) | com.geeksville.analytics.GoogleAnalytics: Analytics: end screen
```

So thanks both so much for the reports and I think I _might_ have fixed it.  If you could test with tonights build that would be great.  If still busted, please open a new bug and at the very least I should now have some new extra log data I asked it to keep.

closed 2020-05-24 by geeksville

feh123, 2020-05-24:
@geeksville thanks for doing this.Do I just download the app from google play now?

geeksville, 2020-05-24:
Sure.  As long as you are an alpha or beta tester. (The non test version
doesn't have there changes).

(Sent from a phone - please ignore typos)

feh123, 2020-05-25:
@geeksville I picked up a beta test version and all looking good this morning. So phone picked up BT's from both radios (from my pairing in settings bluetooth after updating to the beta) and I could see them both in the app. Map showed positions, I could text them and I have a QR icon but channel name and options still ghosted.
Some other info: I see the two radios, one with a green dot, the other an empty circle. I can flip between the two. No matter which one I have the green dot on my text messages go to both. I have tried to name them separately but they both take only one name - both get written at the same time. The radio sleep still happens when the screen goes blank and I get the radio sleeping on my app. 
If I walk around the garden (still under lockdown) with one radio I need to set reset or power off/on to get the screen back from sleep - then it shows the last message - no arrow pointing towards the other radio. Once asleep I have to use reset to get the radios to re-appear in the app. But only the text message appears - how do I get back to the arrow? The app does not record any track I made walking one radio around the garden.
So all really good - thanks for eliminating the bug. Is there a good write up of how to use the app? Apologies for asking so many questions. Thanks again.