meshtastic/Meshtastic-Android #33: autobug on a P9000 in version 0.6.8

geeksville, 2020-06-05:
Thanks for the report @sensoriot.  I'll look into this one early next week.

Non-fatal Exception: java.lang.AssertionError: Assertion failed
       at com.geeksville.android.Logging$DefaultImpls.logAssert(Logging.java:62)
       at com.geeksville.mesh.service.SafeBluetooth.logAssert(SafeBluetooth.java:51)
       at com.geeksville.mesh.service.SafeBluetooth.asyncConnect(SafeBluetooth.java:527)
       at com.geeksville.mesh.service.RadioInterfaceService.startConnect(RadioInterfaceService.java:544)
       at com.geeksville.mesh.service.RadioInterfaceService.retryDueToException(RadioInterfaceService.java:423)
       at com.geeksville.mesh.service.RadioInterfaceService$retryDueToException$1.invokeSuspend(RadioInterfaceService.java)
       at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(BaseContinuationImpl.java:33)
       at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.java:56)
       at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.java:571)
       at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.java:738)
       at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.java:678)
       at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.java:665)

[com.geeksville.mesh_issue_5a69d35a2b8b043053126ff44f0537c6_error_session_5EDA4819026F000108B6257A174C32F5_DNE_0_v2.log](https://github.com/meshtastic/Meshtastic-Android/files/4738010/com.geeksville.mesh_issue_5a69d35a2b8b043053126ff44f0537c6_error_session_5EDA4819026F000108B6257A174C32F5_DNE_0_v2.log)

SensorsIot, 2020-06-05:
I sent it from my old Elephone because it did not connect. This afternoon it ran quite well…

geeksville, 2020-06-05:
@SensorsIot thanks - this was actually the automated report that got sent before you pressed the "report" button and it contained all sorts of great debugging info on the root cause.

geeksville, 2020-06-10:
fixed 0.7.4 - the new non-auto reconnect call used to support old samsung phones broke other things

closed 2020-06-10 by geeksville