meshtastic/Meshtastic-Android #34: Add a settings screen that lets use change any of the RadioConfig parameters

geeksville, 2020-06-05:
i.e. to let them change sleep behavior (possibly put just a gui for "sleep time" in seconds/minutes)

comment from @sensorsiot

geeksville, 2020-07-13:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/a-few-considerations-on-protobufs/729/2

SensorsIot, 2020-09-16:
Hi Kevin,

I hope everything is well on your side. I just wanted to send you a document I got from one of my viewers. If you look at page 9, you see my video about Meshtastic as a strategic weapon. First, Naomi and the Chinese and now the American. Just the Russians are lacking…

I feel like Einstein, who did not like what they did with his discoveries…

Regards

Andreas

andrekir, 2022-02-22:
keeping settings simple and sticking to defaults for now. closing.

closed 2022-02-22 by andrekir