meshtastic/Meshtastic-Android #36: Don't start service if no radio is paired

geeksville, 2020-06-07:
See associated discussion thread:

geeksville, 2020-06-08:
okay - thanks for the report @brianczapiga.  This fix will go out in tomorrow's build (but initially only to alpha/beta testers - If you'd like to join one of those groups you can get it then.  Otherwise it will go to everyone a few days later)

closed 2020-06-08 by geeksville

brianczapiga, 2020-06-08:
Wow, that was really quick. Were you also able to add an exit function? I didn't see one in the referenced commits.

geeksville, 2020-06-08:
yes - though I chose not to call it exit.  In the device list there is now
a "None" device - because I think that is a cleaner UX.  If you select that
the services are stopped so there is no background code (so the app gets
fully suspended like any other 'activity only' app).

commit ce14fde3 (2020-06-08): Fix #36 - allow users to select 'none' for the preferred radio

If none is selected, we do not leave our service running.  And we do
not start it on boot.