meshtastic/Meshtastic-Android #37: Questions about the application?

lewisxhe, 2020-06-08:
I use t-beam.I am using firmware-tbeam-EU865-0.6.7.bin as the test firmware for tbeam， In most cases, I cannot find the Bluetooth device, or when it is found, I connect, after pairing is completed, after using the app, click the input box in the app message box, and the input is not possible. The most seen in the setting bar is "Connected to radio, but it is sleeping". Sometimes after the device has been connected and paired, it still shows in the application that the device needs to be paired in the setting ， Also, I was trying to run this code in TTGO Twatch and encountered the same problem. I suspect this is a problem with the application

lewisxhe, 2020-06-12:
The latest 0.74 firmware has been used. After updating the application, the device cannot be found.

geeksville, 2020-06-12:
hmm - what model of phone is your android device?

lewisxhe, 2020-06-12:
LG V30+

geeksville, 2020-06-12:
Hmn.  Strange.  If the screen is on for the device, and you then go to bluetooth settings on the phone, does the device show up and can you pair with it there?

geeksville, 2020-06-12:
Also: How did you originally install the binaries.  Did you use our device-install.sh script or something else? (because some stock devices sometimes have malformatted system partitions which breaks BLE pairing)

lewisxhe, 2020-06-12:
The device is already paired, but it is not displayed in the application. I use esp download tools to download the firmware. I don’t think it’s a problem with the download tool.

geeksville, 2020-06-12:
So if you go to the (last) settings screen in meshtastic the device is not listed?

lewisxhe, 2020-06-12:
yes

geeksville, 2020-06-12:
hmm - the app is just doing a BLE scan looking for a particular BLE service on our devices.  I haven't heard this report before.  Can you try one more thing: Reboot your LG phone, launch the app, see if the devices show up or not, wait about 20 second and then click "report bug".  That will send a report to Google (if you are in a place that can reach google?) and then I can look at the logs.

lewisxhe, 2020-06-12:
Okay, I found the problem. I didn’t have the permission to turn on the positioning. Generally, when the application requires Bluetooth to be turned on, the positioning is also turned on, but the application has no prompt, so I ignored it. Sorry! I will continue to test!

geeksville, 2020-06-12:
ooh thanks!  That is a bug anyways, because when we ask for location permissions it should have raised a dialog.  I'll repurpose this bug for that purpose.  Thank you for the report.

geeksville, 2020-06-12:
Just checking: Is it possible the app asked once for location permissions but you denied it?  if so android system won't let us ask again.  But we should complain about it.

slavino, 2020-06-12:
I believe there needs to be a separate dialog for BT and level of precision of location access (GPS/GSM network). Checking other stuff may be important as well.  

You may get some inspiration in the https://owntracks.org/ applications.

rradar, 2020-06-12:
Actually what my findings are on various (google free) android (7.1, 8.1) devices is that even if I granted permissions for location the **location service** itself **needs to be turned on** beside bluetooth that the meshtastic app can list any bluetooth devices.

Maybe a special check in the app (like the one which is present to ask to turn bluetooth on if it's off) would do the trick?

geeksville, 2020-06-12:
@rradar Thanks  - yes - it is a requirement to have "fine location" access turned on to use the BLE scanning API (because such info indirectly gives the app some location info).  The app already checks and requests those permissions, but if the user ever says no the app doesn't complain.  I'm adding a reoccurring warning message now.

commit a81edbf6 (2020-06-12): Fix #37, show a Toast every time the app launches about missing permissions
commit e063b463 (2020-06-12): Merge pull request #43 from geeksville/dev

Fix #37, show a Toast every time the app launches about missing permi…
closed 2020-06-12 by geeksville