meshtastic/Meshtastic-Android #50: google crash report with the new channel GUI

geeksville, 2020-06-16:
Hi lgoix,

I just received the following crash report. (Also, if you want - tell me your google friendly username and I'll add you to the google console ;-) ).

If you don't have time to look at it no worries - I'll look into it in the next couple o days.

[com.geeksville.mesh_issue_1d289ec365695d4464c0ed4c8c9bb3f9_crash_session_5EE7E3E0038600017DAD94DF32C6F831_DNE_0_v2.log](https://github.com/meshtastic/Meshtastic-Android/files/4783713/com.geeksville.mesh_issue_1d289ec365695d4464c0ed4c8c9bb3f9_crash_session_5EE7E3E0038600017DAD94DF32C6F831_DNE_0_v2.log)

```
Fatal Exception: java.lang.IllegalArgumentException: Can't get the number of an unknown enum value.
       at com.geeksville.mesh.MeshProtos$ChannelSettings$ModemConfig.getNumber(MeshProtos.java:5811)
       at com.geeksville.mesh.MeshProtos$ChannelSettings.setModemConfig(MeshProtos.java:5950)
       at com.geeksville.mesh.MeshProtos$ChannelSettings.access$10400(MeshProtos.java:5725)
       at com.geeksville.mesh.MeshProtos$ChannelSettings$Builder.setModemConfig(MeshProtos.java:6258)
       at com.geeksville.mesh.ui.ChannelFragment$onViewCreated$2$2.onClick(ChannelFragment.java:171)
       at androidx.appcompat.app.AlertController$ButtonHandler.handleMessage(AlertController.java:167)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loop(Looper.java:193)
       at android.app.ActivityThread.main(ActivityThread.java:6739)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:495)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:859)
```

lgoix, 2020-06-16:
My Google email is the username for all the Google services I use.

ludovic.goix@gmail.com

I will stop the battery display I fix this bug... It's hard to keep the
same quality level as I am in Java with a new language

Ludovic

geeksville, 2020-06-16:
no problem! Just thought I'd mention it because you probably can do it faster better than me

geeksville, 2020-06-16:
btw - I think I just blessed you for crashlytics access:

https://console.firebase.google.com/u/0/project/meshutil/crashlytics/app/android:com.geeksville.mesh/issues

lgoix, 2020-06-16:
there is the fix

geeksville, 2020-06-16:
yay! can you send in a PR and I'll push out a build?

lgoix, 2020-06-16:
I should learn how to make the PR appear here...

geeksville, 2020-06-16:
in the PR just say something like "#50" if you want to mention the bug or "fix #50" if you want to mention the bug and automatically close the bug when the PR gets merged.

geeksville, 2020-06-16:
I'll demo that right now ;-)

lgoix, 2020-06-16:
I think it's because I created the branch after the commit, and just PR that branch

geeksville, 2020-06-16:
you can mention that word either in the commit or the PR, doesn't really matter.

geeksville, 2020-06-16:
oh I see - yeah - gotta mention it in the PR if you want it in the bug

PR #53 (2020-06-16): Dev
fix #50
commit 21e1df80 (2020-06-16): 0.7.76 (hotfix fix #50)
closed 2020-06-16 by geeksville