meshtastic/Meshtastic-Android #51: Unable to connect

thomslik, 2020-06-16:
Did the Report Bug on both phones as you asked.

geeksville, 2020-06-16:
hmm - can you try launching the app again tomorrow (no need to try and debug or press report bug again)?  I don't see any "report bug" reports yet for 0.7.75.  Crashlytics doesn't promise that reports get uploaded immediately (though it usually happens that way), sometimes they don't do it until a number of hours later or the next launch of the app.

lgoix, 2020-06-16:
I am reproducing this while developping and pushing directly the update on my phone.

To fix the issue, I need to :
1) forget the bluetooth device
2) disable bluetooth, enable it after
3) go back to the app and I see the radio available.

thomslik, 2020-06-16:
Tracfone Blu:
T-Beam v1,7.6 bin
Android App 0.7.76
Phone upgraded to 0.7.76.
Cleared cache and data of the app.
Unpaired everything in Bluetooth, Turned it off, then back on. Repaired device.
App just shows None as a choice, you haven’t paired a Meshtastic radio…
Reset T-Beam, same result. Turned it off and restarted, same.
Unpaired and repaired, same.
Sent another bug report.

ZTE Z812 Phone with Android 5.1
Same latest versions of app and bin.
Did all the same resets, etc.
App shows the connected T-beam, but it’s greyed out and not selectable. Also shows the non paired T-Beam as a choice. It wasn’t connectable and later disappeared.
Sent another bug report.

thomslik, 2020-06-16:
Finally got the above ZTE working with the app after numerous clearing caches, bluetooth unpair, off/on repair. T-Beam reset combinations! Still trying with the Tracfone Blu.

thomslik, 2020-06-16:
Tracfone Blu is working now that Location services is enabled at the top level android settings. It was originally only enabled in the app settings. Thanks to geeksville and lgx for solving the problem.

geeksville, 2020-06-16:
per forum thread: problem was location services were disabled.

geeksville, 2020-06-16:
btw @lgoix deserves all the credit on this one ;-)

closed 2020-06-16 by geeksville