meshtastic/Meshtastic-Android #55: warn user if location services are turned off - explains many bluetooth problems

geeksville, 2020-06-16:


geeksville, 2020-06-16:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/alpha-tester-thread-no-new-tests-needed-please-upgrade-to-release-version/298/60

lgoix, 2020-06-16:
The phone needs the location to provides the GPS information. 

The device provide it when it is not paired to the phone ?

geeksville, 2020-06-16:
@lgoix the app only needs that info if the device says it doesn't have a GPS.  If the device has a GPS we don't bother starting the location reading (to save CPU/power on the phone)

geeksville, 2020-06-16:
(and we already had a warning in that case - but only in that case)

lgoix, 2020-06-16:
I will investigate this, yesterday I was debugging for the battery level and wondered why it was using my phone location, there is a GPS, maybe sometime it says it doesn't

commit 03af1f1d (2020-06-16): Fix #55 - warn user if they have location disabled
closed 2020-06-16 by geeksville