meshtastic/Meshtastic-Android #65: Display last time we receive a transmission from a node

lgoix, 2020-06-24:
A date will be displayed in the node list to show last time we have received a communication from this node

geeksville, 2020-06-24:
great!   btw - the lastrx time has the info you'll need (stored in both the node info and in every message)

lgoix, 2020-06-25:
For the battery, almost done

For the last time, there are some problem of timestamp I need to work to fully understand why sometime the value i've got are broken (315967899) when current timestamp is 1593056049. It seems to be a conversion from Long to Int. 

Is there a reason why the time is saved as a Int ?

geeksville, 2020-06-25:
ooh pretty!

re: why is time stored as an int
To keep packet sizes small over the (super slow) lora link I wanted uint32 for the time (as number of seconds since 1970).  But yes, I wouldn't be surprised if somewhere java/kotlin land is having a bug because of that - because of no native uint32 type in java.  When we convert from the MeshProtos.Position into our Position model object we should use longs there (and keep it as as longs in the anndroid app)

closed 2020-07-15 by geeksville