meshtastic/Meshtastic-Android #76: feh123 issue with 0.7.84 and sony

geeksville, 2020-07-01:
per forum

geeksville, 2020-07-01:
hmm - I don't see a report from a Xperia Z1 for 0.7.84 app.  I do see a couple of older reports but I think most of those are  fixed.  The one that was not was this: #77.  So for now I'm going to debug that one.

geeksville, 2020-07-01:
@feh123 what timezone are you in?  I've got a build which fixes one of your problems (not sure if it is what you saw in .84 - no bug report received yet).  But I'd like to put in a few other fixes before I bother pushing something to google play.

geeksville, 2020-07-02:
okay - the thing I thought might be your sony problem (still no bug report for .84 from the googles) is up on the play store 0.7.85 alpha/beta.  Can you let me know how it goes?

feh123, 2020-07-02:
UK time zone. Thanks for debugging.

feh123, 2020-07-02:
@geeksville Just updated my tbeams to 0.7.10 and the app to 0.7.85. App picks up the four tbeams immediately. I go to the Android BT menu and forget the devices there. Then refreshing the BT shows up the devices and I can pair all four tbeams again with no errors. Back in the app I see all four devices but none are connected and the meassage above the devices says "Please pair device in Android Settings". If I go to map or another page and return to device page all the devices are missing. The page only has the None icon ticked. If I go back to Android BT settings, forget the devices and try again they do not pair - I do not see a pin code on the device and eventually get the message incorrect passkey. I reported a bug a this point. If I force stop the app and restart I get back to the start page which shows no devices and says "not connnected, select radio below" (there are none to select). If I turn my phone off and restart I again see the four devices but, as above, although I pair them in the Android BT settings the app still sees them as not connected. At this point I reported a bug again. Sony Xperia Z1 model C6903 running Android 5.5.1. Time zone UK. Sorry for this problem. Just to say I have a tablet (Huawei model CPN-W09,  no sim) running Android 7.0 that also has problems with connecting the devices to the app. It pairs in Android Settings but the App still say please pair - I have reported a bug for that too.

geeksville, 2020-07-02:
re: Sony
Ok - that time the .85 report came through.  here it is:
[com.geeksville.mesh_issue__error_session_5EFD9698026B00015F7F0000DBA52EB9_DNE_0_v2 (1).log](https://github.com/meshtastic/Meshtastic-Android/files/4864978/com.geeksville.mesh_issue__error_session_5EFD9698026B00015F7F0000DBA52EB9_DNE_0_v2.1.log)

Though it looks like the bluetooth scan started "com.geeksville.mesh.ui.BTScanModel: starting scan" but the phone never told us it saw any devices.  Can you try an experiment for me? start the app, see that no devices are listed.  Then while the app is running go to the bluetooth settings and turn off bluetooth, wait 30 seconds and then turn on bluetooth.  Then go back to the app and see if the devices appear?

Also the .84 report came through:
[com.geeksville.mesh_issue__error_session_5EFC7CFC034A0001651A06DB5E1595C6_DNE_2_v2.log](https://github.com/meshtastic/Meshtastic-Android/files/4865008/com.geeksville.mesh_issue__error_session_5EFC7CFC034A0001651A06DB5E1595C6_DNE_2_v2.log)
It shows the app trying to connect but nothing else.  I'll need to investigate

geeksville, 2020-07-02:
Oh!  Duh - Sony was trying to be "clever" (they are not supposed to do this ;-) ) and they filter all log messages that are not "info" priority or above.  So all of the logs we emit that are at DEBUG level are stripped before sending to us.

So I'll add sony to the "badboys club" and we will lie and emit logs at "ERROR" priority so that none get stripped.  This will be in 0.7.86 (which I'll release in a few minutes)

geeksville, 2020-07-02:
also I suspect that this bug (which was in android 5.1.1) is a possible root cause: https://issuetracker.google.com/issues/37121223

I'll try to figure out a work around.   So .80 worked fine for you?  I'll look at what changed and see if it is related.

feh123, 2020-07-02:
Yes with 0.7.80 I was connecting to the tbeams and they would appear in mapbox.

geeksville, 2020-07-02:
ok thanks! - I'll look at the changes since then and see if it relates to that 5.1.1 bug.  In the meantime when you get a chance please try the .86 so I can get logs with more detail

feh123, 2020-07-02:
Just tested app version 0.7.87. I paired in Android settings but the app still said pair device in Android Settings. I reported the bug. I switched the BT off as you said and restarted the app - it asked to allow BT to be switched on. This worked as the app then connected to the radio on pairing! I reported this too. Looking good however - it shows the device name under people and I see the QR. I will try all the tbeams tomorrow. I hope the reports are fine - thanks for helping! Have a good Independence Day.

geeksville, 2020-07-02:
btw - you'll want to update to 0.7.88 (which I just released because I noticed a serious bug in .87)

feh123, 2020-07-03:
@geeksville Great progress. Updated to 0.7.88, forgot all stored devices in BT settings and turned BT off. Opened app, allowed BT and saw all four tbeams. Paired with all. Three appeared on map and were named on people page, with disatnce and battery level. Sent text - it took several minutes but text arrived at all four devices. The one missing from the map had ??? instead of the name that appeared on the other three. I sent report bug. The one missing shows the not connect cloud icon. Is there a limit on connected devices? Will keep on testing. Thanks!

feh123, 2020-07-03:
@geeksville An update. I left the devices, switched on, most of the day. The App loses connection - I see either the radio is asleep or please pair again. It seems that the only way to re-establish is to completely pair again from an empty BT Android Settings.  I reported a bug. I like the new additions to the device - the ability of change the brightness of the screen and all the extra info on gps signal (what does Exc mean) and battery level.

geeksville, 2020-07-04:
yay!  got a good log from that run now that I stopped sony from filtering.  I'll look into this this morning

[com.geeksville.mesh_issue__error_session_5EFEDDDD03CB00014B4EFFD0004052D7_DNE_2_v2.log](https://github.com/meshtastic/Meshtastic-Android/files/4873345/com.geeksville.mesh_issue__error_session_5EFEDDDD03CB00014B4EFFD0004052D7_DNE_2_v2.log)

geeksville, 2020-07-04:
not sure if this was the bug for you (though it was busted on your phone), but a fix is in for 0.7.89.  Can you let me know if it helps?

geeksville, 2020-07-05:
@feh123 okay one more change as we remote debug the problem with this app on android 5.1.1.  Can you install 0.7.90 and let me know:

1) Does it work without having to toggle bluetooth on and off?
2) If it doesn't work, while the app is running go toggle bluetooth off, wait 30 secs, toggle back on and then see if the app can connect?

If step 2 fails click report bug and I'll fix/debug ;-).  Thanks for helping with this.  I don't have a 5.1.1 device with BLE.

feh123, 2020-07-05:
@geeksville thanks for work! I used two tbeams, that had gps fixes running 0.7.10 with the App on 0.7.90. BT was on and I had not cleared any devices. Started App and saw two tbeams immediately. Received paring request and seemed to pair. Icon said disconnected but I could see the tbeam on the map. However I could not text it. I had problems with the second tbeam - not seen on the map and soon disappeared from the device list on the App. I did a RB at this point.
I toggled BT off and used App to turn it on. Much better - the two paired and I see them on the map. I can text them although it is slow to deliver the message (over 5 minutes). I did another RB here. Added third tbeam and that worked and I could text all three - much faster this time. I still get the issue that tbeams disappear from the map and I get the not connected icon. So good news that I can see all three on the map. Hope the RB's help - glad to help fix a slow old Sony. Unfortunately I updated to an iPhone.

geeksville, 2020-07-07:
thanks - that sounds useful.  I'll wait for the reports to come in. (not here yet - sometimes takes a couple of days)

The map position update bug is fixed and will be in the next release

closed 2020-07-08 by geeksville