meshtastic/Meshtastic-Android #80: Samsung S5 (SM-G900F) Android 6.0.1

slavino, 2020-07-02:
[image]

Device fails to display QR code and channel name

geeksville, 2020-07-02:
interesting!  It looks like it might screen size scaling issue.  Otherwise the device basically works okay?  I can make a similar screen in the android sim and try to reproduce.

geeksville, 2020-07-02:
hmm - no the screen scaling is not a problem, I just tried it in sim with a bunch of screen sizes.  hmm.

geeksville, 2020-07-02:
hmm - the problem is that channel is null (when it should not be).  To help me figure this out can you do two things:

* Does the channel QRcode show up while the device is connected?
* After the channel is not showing can you click "report bug" in the app and then make a note here? and I'll go look at the logs.

slavino, 2020-07-03:
Let's check once I have my boards T-Beam_v0.7 flashed with production build from GITHUB releases.

Could be really only some constellation of TBeam_v1.x firmware on TBeam_v0.7 hardware. Power management behavior differs quite a lot between these two as on older board is AXP202X missing...

geeksville, 2020-09-16:
I fixed some screen scaling.  I think this is fixed now.  If not, would you mind reopening?

closed 2020-09-16 by geeksville