meshtastic/Meshtastic-Android #96: It is necessary to connect the buzzer to device incoming messages.

luxonn, 2020-07-12:
and the application also needs sound

geeksville, 2020-07-17:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/notification-and-beacon-mode/768/2

Commander-Crash, 2020-07-17:
It looks like the team is on it for the Notifications on new msg. Thanks for your work :)

geeksville, 2020-08-10:
cc @lgoix

geeksville, 2021-01-21:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/android-app-new-releases/78/102

mc-hamster, 2021-11-18:
andrekir — Today at 11:05 AM
great!
https://github.com/meshtastic/Meshtastic-Android/issues/96 (Note sounds OK)
https://github.com/meshtastic/Meshtastic-Android/issues/154 (date stamp OK)
https://github.com/meshtastic/Meshtastic-Android/issues/185 (Note sounds OK)
https://github.com/meshtastic/Meshtastic-Android/issues/194 (elapsed minutes OK)
https://github.com/meshtastic/Meshtastic-Android/issues/258 (Dark theme OK)
https://github.com/meshtastic/Meshtastic-Android/issues/279 (User reports resolved)
https://github.com/meshtastic/Meshtastic-Android/issues/111 (Solved in PR #234)
https://github.com/meshtastic/Meshtastic-Android/issues/281 (Solved on 1.2.45)
https://github.com/meshtastic/Meshtastic-Android/issues/289 (Solved on 1.2.45)

closed 2021-11-18 by mc-hamster