meshtastic/Meshtastic-Android #111: In russian the "allow analytics" checkbox looks really bad and covers most of the screen

geeksville, 2020-07-22:
https://aws1.discourse-cdn.com/free1/uploads/meshtastic/optimized/1X/3dc2115f22f3716753a6f735efb73fd341d22247_2_243x500.jpeg 
see linked thread...

geeksville, 2020-07-22:
(the layout should be smarter about max width for the "submit bug" button and should allow the other text to flow below it?

mc-hamster, 2021-11-18:
andrekir — Today at 11:05 AM
great!
https://github.com/meshtastic/Meshtastic-Android/issues/96 (Note sounds OK)
https://github.com/meshtastic/Meshtastic-Android/issues/154 (date stamp OK)
https://github.com/meshtastic/Meshtastic-Android/issues/185 (Note sounds OK)
https://github.com/meshtastic/Meshtastic-Android/issues/194 (elapsed minutes OK)
https://github.com/meshtastic/Meshtastic-Android/issues/258 (Dark theme OK)
https://github.com/meshtastic/Meshtastic-Android/issues/279 (User reports resolved)
https://github.com/meshtastic/Meshtastic-Android/issues/111 (Solved in PR #234)
https://github.com/meshtastic/Meshtastic-Android/issues/281 (Solved on 1.2.45)
https://github.com/meshtastic/Meshtastic-Android/issues/289 (Solved on 1.2.45)

closed 2021-11-18 by mc-hamster