meshtastic/Meshtastic-Android #113: autobug on android 8.1 - crash due to google bug in JobIntentService

geeksville, 2020-07-25:
google bug here: https://issuetracker.google.com/issues/63622293

Possibly try workaround mentioned in comment 110 but see comment 115 for caveat.

crashlytics report here: https://console.firebase.google.com/u/0/project/meshutil/crashlytics/app/android:com.geeksville.mesh/issues/dd9cd2d1988c6370a955c38f29ef9510?time=last-seven-days&sessionId=5F1B6668018E00010231E605344DC5BC_DNE_0_v2

Caused by java.lang.SecurityException: Caller no longer running, last stopped +306ms because: timed out while starting
       at android.os.Parcel.readException(Parcel.java:2005)
       at android.os.Parcel.readException(Parcel.java:1951)
       at android.app.job.IJobCallback$Stub$Proxy.dequeueWork(IJobCallback.java:191)
       at android.app.job.JobParameters.dequeueWork(JobParameters.java:208)
       at androidx.core.app.JobIntentService$JobServiceEngineImpl.dequeueWork(JobIntentService.java:315)
       at androidx.core.app.JobIntentService.dequeueWork(JobIntentService.java:640)
       at androidx.core.app.JobIntentService$CommandProcessor.doInBackground(JobIntentService.java:390)
       at androidx.core.app.JobIntentService$CommandProcessor.doInBackground(JobIntentService.java:383)
       at android.os.AsyncTask$2.call(AsyncTask.java:333)
       at java.util.concurrent.FutureTask.run(FutureTask.java:266)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
       at java.lang.Thread.run(Thread.java:764)

geeksville, 2020-07-25:
alas - whatever JobIntentService this is, it isn't created by us (directly) - so can't apply the workaround.

sachaw, 2021-12-29:
Is this still happening? seems very old.

andrekir, 2021-12-29:
no issues in Crashlytics since. closing.

closed 2021-12-29 by andrekir