meshtastic/Meshtastic-Android #120: BT and Channle issue

Commander-Crash, 2020-07-30:
My info:
2x LILYGO ESP32 LoRa OLED TTGO V.1
firmware-tlora-v1-US-0.9.1.bin
Android TCL A50DL V.8.1.0
2nd Android Platinum_B5 V.8.1.0
Both are running  Meshtastic 0.9.01 
__________________________________________

Ok Let me Begin at I know its beta I'm not complaining.This is so others that have this issue may find a solution.
I don't have a top of the line flagship phone However after upgrading from 7.11FW to 0.9.1FW Issue with Bluetooth not wanting to reconnect when radio is out of sleep. Bluetooth will disconnect when LoRa radio is put to sleep. I want to run these nodes 24/7 I don't care for the radio to go to sleep I have the power to support them being on 24/7 I know that the device is not to blame due to androids terrible handling on Bluetooth profiles. I tried to switch back to 7.11FW and at 1st no issue but On the Android Platinum_B5 Bluetooth will not reconnect after radio wakes up. Android TCL A50DL has no issues with bluetooth.
Now the Channel name. I never got this to work at all It only works if both channel name is set to "Default" when changing the name to like "29a" or "MikesNode" It will not see the other device even with same channel name. I would like to be able to secure my chat. Like I said I don't know if I need a flagship phone for all this to work well. I had no issues with Bluetooth until recently. I have tried other phones and only 2 I own work somewhat ok with meshtastic.

Whatever info you may need just ask. Thanks.
[image]

geeksville, 2020-08-10:
good point - we need an easier way to let users say "don't have the device go to sleep"  but in the meantime here's a workaround you can use.

https://meshtastic.discourse.group/t/would-you-like-your-devices-ble-link-to-not-go-to-sleep-do-this/552/2

geeksville, 2020-09-17:
closing as a dup of #161 (essentially same request)

closed 2020-09-17 by geeksville