meshtastic/Meshtastic-Android #137: Device keeps disconnecting from phone

luandro, 2020-08-24:
**Describe the bug**

Having a hard time pairing a Meshtastic device to phone. I've been able to connect in the past, but also with some difficulty. Been trying for a few hours now without success.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on the Meshtastic device (T-Beam)
2. On Meshtastic Android app check that the device shows up and pair with it
4. After pairing and showing up on the app there's a Bluetooth notification "Disconnected"
5. On the T-Beam's serial monitor there's this log:
```
PASSKEY_ACTION_EVENT started 
*** Enter passkey 897248 on the peer side ***
Trigger powerFSM 7
Transition powerFSM transition=Bluetooth pairing, from=DARK to=ON
ble_sm_inject_io result: 0
connection updated; status=0 handle=0 our_ota_addr_type=0 our_ota_addr=01:XX:XX:ab:fa:10
 our_id_addr_type=0 our_id_addr=01:XX:XX:ab:fa:10
 peer_ota_addr_type=1 peer_ota_addr=00:79:67:c0:0d:ed
 peer_id_addr_type=1 peer_id_addr=00:79:67:c0:0d:ed
 conn_itvl=6 conn_latency=0 supervision_timeout=2000 encrypted=0 authenticated=0 bonded=0

encryption change event; status=0 handle=0 our_ota_addr_type=0 our_ota_addr=01:XX:XX:ab:fa:10
 our_id_addr_type=0 our_id_addr=01:XX:XX:ab:fa:10
 peer_ota_addr_type=1 peer_ota_addr=00:79:67:c0:0d:ed
 peer_id_addr_type=1 peer_id_addr=00:79:67:c0:0d:ed
 conn_itvl=6 conn_latency=0 supervision_timeout=2000 encrypted=1 authenticated=1 bonded=1

connection updated; status=0 handle=0 our_ota_addr_type=0 our_ota_addr=00:XX:XX:ab:fa:10
 our_id_addr_type=0 our_id_addr=00:XX:XX:ab:fa:10
 peer_ota_addr_type=0 peer_ota_addr=00:a4:XX:d6:1f:XX
 peer_id_addr_type=0 peer_id_addr=00:a4:XX:d6:1f:XX
 conn_itvl=39 conn_latency=0 supervision_timeout=2000 encrypted=1 authenticated=1 bonded=1

GPS fix type 3
Update DB node 0xabfa10d8, rx_time=1598286305
Node status update: 1 online, 2 total
New GPS pos lat=-20.0000, lon=-40.0000, alt=891, pdop=1.990000, heading=27790.964850, sats=7
disconnect; reason=531 handle=0 our_ota_addr_type=0 our_ota_addr=00:XX:XX:ab:fa:10
 our_id_addr_type=0 our_id_addr=00:XX:XX:ab:fa:10
 peer_ota_addr_type=0 peer_ota_addr=00:a4:XX:d6:1f:XX
 peer_id_addr_type=0 peer_id_addr=00:a4:XX:d6:1f:XX
 conn_itvl=39 conn_latency=0 supervision_timeout=2000 encrypted=1 authenticated=1 bonded=1
```

**Expected behavior**
Expected device to connect to phone and work.

**Screenshots**
[image]

**Device info:**
 - Device model: TBEAM
 - Software Version 0.9.1

**Smartphone information (if relevant):**
 - Device: Motorola G3
 - OS: Android 6
 - App Version 0.9.03

geeksville, 2020-08-24:
oops sorry about that. 

Two ideas:

* Starting around 0.9 we switched to a much nicer (but completely different) BLE stack - because the old one had a would-not-fix bug that could forget pairing.  If your prior device software version was <0.9 could you try on the phone: forget the old device in bluetooth settings.  Then in bluetooth settings then re-pair with the phone.

* If that doesn't work, after pairing:
1) launch the app
2) press the reset button on the device (to force it to reconnect)
3) note any change in the GUI (showing gaining/losing connection for instance)
4) wait 10ish seconds (so the logs chill out) then click "report bug"
5) normally the android bug reporter will wait up to two days to upload that bug report.  To force it to upload promptly, force quit the app and then relaunch it.  At launch (behind the scenes) the app will upload the bug report.

Add a note to this bug and I'll go find the report by your phone model.

luandro, 2020-08-25:
Thanks @geeksville, this project is amazing btw.

Pressing the reset button actually got it working. I'll keep this debug process in mind, and wil be sending bug reports in case it keeps happening.

luandro, 2020-08-25:
Trying on another phone, same model (Moto G3). Having exact same problem. Resetting the T-Beam isn't solving.

Using `adb logcat` to find out more:
```
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: com.geeksville.mesh.service.RadioNotConnectedException: We don't yet have our myNodeInfo
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at com.geeksville.mesh.service.MeshService.getMyNodeNum(MeshService.kt:3)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at com.geeksville.mesh.service.MeshService.access$getMyNodeID$p(MeshService.kt:1)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at com.geeksville.mesh.service.MeshService$binder$1$getMyId$1.invoke(MeshService.kt:1)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at androidx.appcompat.app.AppCompatDelegateImpl$ConfigurationImplApi17.toRemoteExceptions(AppCompatDelegateImpl.java:1)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at com.geeksville.mesh.service.MeshService$binder$1.getMyId(MeshService.kt:1)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at com.geeksville.mesh.MainActivity.updateNodesFromDevice(MainActivity.kt:19)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at com.geeksville.mesh.MainActivity$mesh$1$onConnected$1.invoke(MainActivity.kt:34)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at androidx.appcompat.app.AppCompatDelegateImpl$ConfigurationImplApi17.exceptionReporter(AppCompatDelegateImpl.java:1)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at com.geeksville.mesh.MainActivity$mesh$1.onConnected(MainActivity.kt:3)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at com.geeksville.android.ServiceClient$connection$1$onServiceConnected$1.invoke(ServiceClient.kt:7)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at androidx.appcompat.app.AppCompatDelegateImpl$ConfigurationImplApi17.exceptionReporter(AppCompatDelegateImpl.java:1)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at com.geeksville.android.ServiceClient$connection$1.onServiceConnected(ServiceClient.kt:1)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at android.app.LoadedApk$ServiceDispatcher.doConnected(LoadedApk.java:1223)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at android.app.LoadedApk$ServiceDispatcher$RunConnection.run(LoadedApk.java:1240)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at android.os.Handler.handleCallback(Handler.java:746)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at android.os.Handler.dispatchMessage(Handler.java:95)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at android.os.Looper.loop(Looper.java:148)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at android.app.ActivityThread.main(ActivityThread.java:5443)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at java.lang.reflect.Method.invoke(Native Method)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:728)
08-25 09:32:52.552 31271 31271 E toRemoteExceptions: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:618)
```

geeksville, 2020-09-10:
I think I just encountered this bug on one of my devices.  It was related to a bug with how NimBLE copes with preferences filling up. 

Fix here: https://github.com/meshtastic/Meshtastic-device/issues/370

I'll release a new bin today or tomorrow.  When I do can you try it and report back here?

luandro, 2020-09-11:
That's awesome @geeksville! Yes, will keep my eye on the releases.

Luckily this only happened on first pairing, after that I never had any problems again.

closed 2021-01-05 by geeksville