meshtastic/Meshtastic-Android #140: Feature Request: Make QR code larger

mc-hamster, 2020-08-28:
Feature request:

Please make the QR code in the Android App larger. The problem is, on my inexpensive test Android devices, the camera doesn't have enough resolution to read the QR code, even by moving it closer.

My current work around is to take a picture of the Meshtastic QR code using my iPhone, enlarge the picture on my iPhone and then use the QR code reader on the Android device to read the QR code on the iPhone.

Proposed solution would be that if the QR code is clicked, the QR code goes into full screen mode.

geeksville, 2020-08-29:
what was this model of phone again btw? I'll plug it into the emulator and take a look.

geeksville, 2020-08-30:
never mind - found it

mc-hamster, 2020-08-30:
This is the Blu View 1 running Android 9.

https://www.bestbuy.com/site/simple-mobile-blu-view-1-black/6373977.p?skuId=6373977

The Best Buy on Harrison St in the SF has them in stock. They're fairly capable devices. If you would like to borrow mine for any reason, I think I live about 20 minutes from you.

mc-hamster, 2020-08-30:
You responded right before I clicked "Comment", right ton.

geeksville, 2020-08-30:
@mc-hamster heh!  I might take you up on that offer for the other more tricky bugs (if not quickly apparent from the autobug report)

mc-hamster, 2020-08-30:
Whatever makes it easy.

closed 2020-08-30 by geeksville

mc-hamster, 2020-08-30:
fixed. thanks!

commit a985f1f2 (2020-08-30): Fix #140 - make QR code larger (by making layout agnostic to display size)