meshtastic/Meshtastic-Android #141: ###Default names with multiple # symbols

moxy-bc, 2020-08-28:
Using the Android app (v0.9.02) with TTGO Meshtastic 0.9.1 creates channel names with multiple ### signs.  It is very hard to set it to a channel and have it stick.  After multiple reboots I finally got it to change, but there's a bug in the handling of names.  Not sure if it's in the Android App or the Meshtastic firmware
[image]

geeksville, 2020-08-29:
ooh that sounds bad - thanks.

geeksville, 2020-08-30:
alas - I can't figure out how to reproduce this (though obviously I believe your picture).  I'm releasing a new device load tonight, would you mind trying it and see if still shows this for you?

dzaczek, 2020-09-08:
I have this same issue . It happen when i try change channel to open Default  then and wait few minutes and a aplication  asking me for upodate information after that i see the network i closed and name is "###Default-" . To simulate open network and change name to Default then restart esp32
[image]
[image]

geeksville, 2020-09-08:
ooh thank you @dzaczek for hte great steps to repro.  I'm sure that will help a lot.

geeksville, 2020-09-08:
problem turned out to be accidentally running UI handler code when the device disconnects/reconnects.

commit e07847af (2020-09-08): fix #141 don't let extra # into channel name. thx @moxy-bc and @dzaczek
closed 2020-09-08 by geeksville