meshtastic/Meshtastic-Android #142: Devices not detected.

PCRobert, 2020-08-28:
Mestastic Android Ver. 0.9.04 (Google Play)

"Please pair device in Android Settings" is displayed with only "None (disable)" only available radio button even when Bluetooth pairing is in place. No amount of killing the app, restarting the TTGO devices, reinstalling the app, reestablishing Bluetooth pairing, or restarting the phone/tablet in any combination seems to make a difference. 

Replicates on Android devices:
LG V30+/Android 9/Kernel 4.4.153
Samsung Galaxy S8/Android 9/Kernel 4.4.111-17263988
Dragon Touch Y88X_PRO(Kids Tablet)/Android 9/Kernel 4.4.167

Replicates with Meshtastic devices:
2x T-BEAM T22 V1.0
Firmwares tested - binary releases of 0.9.3 & 0.9.1 (firmware-tbeam-US-0.9.3.bin, firmware-tbeam-US-0.9.1.bin)

I'm not sure how else to describe the problem or what other information to provide. Please let me know what else might be helpful.

geeksville, 2020-08-29:
hmm - do you also have location permissions enabled for the app?

PCRobert, 2020-08-29:
In all three cases, yes, and to the highest level available in each interface.

PCRobert, 2020-08-31:
Some progress -- on both the LG V30+ and the Galaxy S8, Meshtastic Android *will* pick up the BT pairing, but it takes over 20 minutes in each case. On the V30, I discovered this by accident of making some more attempts of re-installing the app, rebooting the phone, etc. and then putting in active time on the phone for a longish period only to switch back to the app and find it working as expected.

Replicates on the S8. On freshly pairing, to a T-BEAM, it takes at least 20 minutes of the Mestastic app being open before any devices are listed in the configuration screen. 

No amount of waiting or messing about works on the Y88X_PRO tablet, but that's sort of excusable for an off-brand cheap kids tablet.

On the previous question, I'm no Android developer, but why on Earth would the location services permission have anything to do with listing paired Bluetooth devices?!? The implication seems rather insane when thought about for a moment.

Is there anything else I can do to help diagnose? Thanks!

geeksville, 2020-09-02:
hmm - I've never heard of reports of such a long delay for pairing.  I'll think about it and come back to this bug.

re "On the previous question, I'm no Android developer, but why on Earth would the location services permission have anything to do with listing paired Bluetooth devices?!? The implication seems rather insane when thought about for a moment."

The reason is that Android is pretty privacy sensitive and with the invention of BLE beacons out in the world (i.e. from Facebook physical tracking group etc), letting apps see BLE devices (especially because you can get implicit/explicit signal level info) can "leak" location information into apps.  So they are trying to be transparent to users about that. (I'm a former Android eng so I remember the drama/pain ;-) )

geeksville, 2020-09-16:
Hi @PCRobert - can you try the current app (0.9.90 or later).  I think this should be fixed now.

PCRobert, 2020-09-16:
So some pretty good improvement. Updating from 0.9.04 to 0.9.90 on the V30 and freshly pairing took just over 10 minutes for a device to be listed. Full-uninstall/re-install and it took 55 seconds for a device to be listed after pressing the "Allow Location Access" button. I'm not sure what difference exists between update and clean install, but that was the only variable. 

I should be able to try on the S8 later today. Are further results of interest?

geeksville, 2020-09-18:
hmm that's interesting.  no need to test the S8 because there are a fair number of successful runs on that phone with these new changes.  thanks for your help!

closed 2020-09-18 by geeksville