meshtastic/Meshtastic-Android #143: "You must turn on location services..." even though it is turned on.

mc-hamster, 2020-08-29:
On a Samsung Galaxy S6 with 0.9.05, I get the error "You must turn on location services in android settings" even though the location permission is enabled in Android settings.

I used the report bug feature in the app Aug 29 @ 12:30PM.

Device Model: SM-g920V
Android: 7.0

Let me know if you need anything else.

geeksville, 2020-08-29:
ok thanks - I'll look for the bug tonight.  Just to double check, a) you have location services enabled in the phone global settings and in the app permissions you've granted permission to the app to access location at all times?

mc-hamster, 2020-08-29:
I dug into it a bit and figured it out.

The Samsung Galaxy S6 has three options for "Locating method" under the location property screen in settings:

- High Accuracy ... Uses GPS, Wi-Fi, and mobile networks to estimate your location.
- Battery Saving ... Uses Wi-Fi and mobile networks to estimate your location
- Phone Only ... Uses GPS to estimate your location.

Battery saving is the manufacture default (this device was factory reset yesterday). In both "Battery saving" and "Phone only", the Meshtastic Android app will return the error message.

When "High accuracy" is selected, I don't get the location error message.

geeksville, 2020-08-30:
ok - can you check to see if this fixes the problem of devices not showing in the settings screen?  Thanks for this (I'll change the warning to tell user they need to enable High Accuracy access)

mc-hamster, 2020-08-30:
This does not fix the problem of devices not showing in the settings screen.

closed 2020-08-30 by geeksville

mc-hamster, 2020-08-30:
Confirmed that I see the new text. Thanks!

commit 502ad2c4 (2020-08-30): Fix #143 tell user we need high-accuracy access (Samsung apparently lets them choose)