meshtastic/Meshtastic-Android #144: Meshtastic Android crashed on pairing with Meshtastic Device

mc-hamster, 2020-08-29:
Meshtastic Android App  0.9.05 crashed when pairing to a Meshtastic device running 0.9.3.

I used the report bug button in app at about Aug 29 @ 12:43PM.

This is on a Blu View 1 on Android 9.

geeksville, 2020-08-29:
ok great - I'll look for hte log report tonight!

geeksville, 2020-08-30:
hmm - these logs look approximately correct (found devices and listed them and seemed to be talking to the device).  I'll look more in depth later tonight.  But just to move them into the bug:
[bluview3.log](https://github.com/meshtastic/Meshtastic-Android/files/5147349/bluview3.log)
[bluview2.log](https://github.com/meshtastic/Meshtastic-Android/files/5147350/bluview2.log)
[bluview1_com.geeksville.mesh_issue_4518e386cd19614e403892a89decaac6_error_session_5F4B03DD02340001262141FCCE1E32E0_DNE_0_v2.log](https://github.com/meshtastic/Meshtastic-Android/files/5147351/bluview1_com.geeksville.mesh_issue_4518e386cd19614e403892a89decaac6_error_session_5F4B03DD02340001262141FCCE1E32E0_DNE_0_v2.log)

geeksville, 2020-09-08:
@mc-hamster  hmm - I'm still having a super hard time reproducing this.  Was this with your own build or with a standard prebuilt bin?

mc-hamster, 2020-09-09:
This was with a standard build.

mc-hamster, 2020-09-09:
I'll do a round of testing tomorrow or Thursday with the updated binaries and android app to reproduce this again.

mc-hamster, 2020-09-11:
I have not been able to reproduce this again using the latest Android app (0.9.80).

Will keep using it over the weekend. If this doesn't happen again by then, I'll close this bug.

mc-hamster, 2020-09-12:
Closed. I can't reproduce this with 0.9.80 or 0.9.81.

closed 2020-09-12 by mc-hamster