meshtastic/Meshtastic-Android #145: Signal and battery indicator for powered-off Tbeam

DDVanMS2018, 2020-08-29:
In the list of units for the App, the time is behaving like it's connected, but the tbeam is off.  The map is fine, the powered-off unit doesn't display.  Is this normal behavior?  I think it should show a "?" for a unit that is off or unreachable.  My first issue post, be kind.  

Thanks.

App v0.9.05
tbeam 1.0-US/0.9.3
T-Beam V1.1 w/ NEO-6M
T22_V1.1 20191212

DDVanMS2018, 2020-08-30:
I also encounter the "?" for the device that is on and connected via BT.  I get the time for the device that is off.  

The notification often says "Conected: 2 of 2 online" when I only have 1 powered-on.

lgoix, 2020-09-17:
@geeksville if devices is powered-off, why the phone is still receiving its infos ?

andrekir, 2021-12-16:
this should be resolved. if you find this error still occurs in recent app & device firmware versions please reopen this issue.

closed 2021-12-16 by andrekir