meshtastic/Meshtastic-Android #147: No available bluetooth devices found on "Select radio below" list.

mc-hamster, 2020-08-30:
On a Samsung Galaxy S2 with Android 7 with the Meshtastic Android App v0.9.05, none of my mesh devices show on the list of available devices within the app but they do show up when I go into bluetooth settings.

I clicked the report bug button at Aug 29 at around 5:17pm.

Screen shots:

[image]
[image]

geeksville, 2020-08-30:
Interesting that the devices are showing up in the system BLE scan screen but not the app screen.  Did you grant the app location access? (without that Android doesn't allow apps to see BLE devices)

mc-hamster, 2020-08-30:
Yes, app location access is granted.

[image]

geeksville, 2020-08-30:
hmm - I haven't received a report yet from a samsung galaxy yet.  Can you try rebooting your phone and running meshtastic (the google bug reporter thingy may wait up to 2 days otherwise)

mc-hamster, 2020-08-30:
Resent @ 1:48pm

lgoix, 2020-09-03:
I'm having the issue, just switched to another tab in the app and back to setting make the devices appear

mc-hamster, 2020-09-03:
> I'm having the issue, just switched to another tab in the app and back to setting make the devices appear

I tried that on my end, no go.

mc-hamster, 2020-09-14:
Everything seems to work as expected.

closed 2020-09-14 by mc-hamster