meshtastic/Meshtastic-Android #153: Feature Request: Display node details in maps when a node is tapped (clicked on)

DingoOz, 2020-09-06:
Currently the maps screen shows the name and a standard icon for each radio in the current channel (mesh).

This Feature Request is for additional details to be displayed for a single node (radio) when the user clicks on that node.

For example three nodes might be displayed on a screen using current UI/UX behavior (Alpha, Bravo and Charlie). Then a user taps on Charlie and the additional details are drawn on screen under the node's name. Additional details might include:
* last time that node was visible to the user;
* distance from user's node;
* current compass heading from user's node; and
* Lat/log of the node 'Charlie'.

To avoid cluttering the screen, these additional details could be displayed until:
* the user clicks the screen again;
* a timer of 30 seconds; or
* the user clicks that specific node again (toggle)

geeksville, 2020-09-07:
this is a great idea (and pretty easy because we can hand off most of this goo to mapbox - we just didn't bother doing it at first)

DingoOz, 2020-09-08:
Thanks. I am going to work on this Feature Request

geeksville, 2020-09-08:
oh that would be awesome!  Let me or @lgoix if you encounter any problems getting the dev environment setup.

DingoOz, 2020-09-09:
Thanks. Dev environment all set up and working thus far. I've added some notes re MapBox in the Meshtastic-Android wiki.

I am working through this Feature Request. No blockers as yet. 

I am hoping to be able to follow MapBox best practices and have the additional information displayed in a callout box when clicked, ...then click again to stop displaying.

I am not really a UI/UX person so hopefully this is the expected behaviour.

geeksville, 2020-09-10:
that sounds amazing.  yah, I'm not really a UI/UX person either. :-)

closed 2022-08-30 by andrekir