meshtastic/Meshtastic-Android #159: Respect Landscape and Portrait screen rotation

nconder, 2020-09-14:
I have a landscape by default phone ([Gemini by Planet Computers](https://store.planetcom.co.uk/collections/devices/products/gemini-pda-1)) and Metastatic will not present its self in portrait mode.  I have even tried several apps that will try to force apps into landscape or portrait.  However, I noticed that for a brief moment when you wake the phone up it will flash to portrait mode then back again to landscape.  I do not have this issue with any other apps on the is phone.  I'm not sure how I can provide you with more information as the app does not crash.  Please let me know if I can supply you with anything.

Android 8.1.0

geeksville, 2020-09-15:
good idea.  Also I've been wanting to make a tablet layout that has the map view and the chat view simultaneously visible.

geeksville, 2022-07-12:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/android-app-x-x-xx/6129/2

qjayjayp, 2023-10-22:
I'm up voting on this too, my use case is a chest rig where you mount your phone on your chest (for hiking etc.) and you can flip it down to have a look and operate it (this by design of the rig is in landscape mode, so it's a bit awkward in practise) and flip it back up when not using it. I was actually surprised to learn that landscape mode is not a OS enforced function but a app decision, would be great if that can be enabled (normally I don't see apps do a whole lot of fancy re-layouting, just stuff get wider and you may scroll more to get down to the bottom of a screen).

nconder, 2023-10-22:
I have had a Gemini for years to and hated that I cannot use with meshtastic thanks for opening this ticket.

caveman99, 2023-11-17:
can you test the APK from this build? https://github.com/meshtastic/Meshtastic-Android/pull/782

qjayjayp, 2023-11-17:
Hi caveman99,

thanks! It works as expected, definitely usable for me and better than portrait.

I'm not using the QR code (I do all programming via USB initially), but for me the QR code does not fit in one piece on the screen anymore (haven't tried it, but I guess it's not usable), this is probably what I guess quite some people are using (as it's easy).

Not so important points but worth a feedback are the following:

- The title line (with the Meshtastic logo), uses up a lot of wasted top/bottom spacing which would give more room for the pages if lowered.
- On a message page I can only see the keyboard and the text window but there is no more space for messages, this is probably nothing the app can do something about (beside the above mentioned point). For me this is working via closing the keyboard with the 'back' key that my phone shows as the standard android set of soft-keys on the right/bottom side (back, home, running apps overview). So I type the message, send it and then close the keyboard, to see 3 one line messages (if further changes will be done a point here could be to change the message layout, not to be text above time and send status, but maybe next to each other so more messages fit on the screen).
- Maybe a specialty of my test phone (Xiami Redmi 9A) is that it has the selfie camera (the one on the screen side) inside the screen (on the top in the middle), so it uses up a bit of screen space that splits the top status bar of android in half in portrait mode. When in landscape mode, this splitted space is not used (by any app), and normally becomes black (which blends in with the frame, so the loss in space is not noted), in this app it's white, which makes the dent the camera produces visible, and also the android status bar in landscape mode extends in this splitted space (ok i finally made some pictures, so it's easier to understand, though it's the least important thing as it's only a visual difference to other apps).
[image]
[image]

Thanks for the change! I have only one last question, is there something known why I should avoid using this build, e.g. due to other ongoing changes? (for the moment is working perfectly fine)

caveman99, 2023-11-17:
the problem with the smaller screen in portrait mode is, that i don't have this phone to test. I made this patch with a Oneplus Pad which obviously has way more screen estate and looks fine, including the qr code. 

I'd like to call this one 'tablet mode' rather than 'landscape mode' because of the shortcoming on smaller displays you noticed. One point worth investigating: My phone can 'undock' the keyboard in landscape mode so the screen is not compressed any more. The Barcode scanner will force portrait mode regardless, because it won't work if you try to use it in landscape.

caveman99, 2023-11-17:
Also: if this is approved, it will be part of the next regular release. The debug APK you're running right now should do fine till then.

qjayjayp, 2023-11-17:
> One point worth investigating: My phone can 'undock' the keyboard in landscape mode so the screen is not compressed any more.

Ah yes, this works for me too, the keyboard gets way smaller (actually I have to check whether it's still practical when on the go) and is slightly transparent, but mainly it can be moved to the empty space on the top left in message pages.

closed 2023-11-18 by andrekir