meshtastic/Meshtastic-Android #166: Will not run de-Googled devices

nconder, 2020-09-18:
Will not run de-Googled devices or devices with out Google Play. The app instantly crashes. 
Remove Google dependencies and also publish alternate stores like F-Droid.

Nexus 6 
LineageOS
Android 7.1.2

geeksville, 2020-09-27:
Hmm - I just tested on the emulator with a bare android SDK (no google play installed).  Seemed to work fine for me.  Can you provide an exception trace or logcat output and I'll happily fix.

geeksville, 2020-10-08:
alas - didn't hear back from reporter.  Without a stack trace or logcat output (because we can't get analytics for that fault) I can't do much to make it work on your device.  @nconder if you can provide that info I'm happy to debug.

closed 2020-10-08 by geeksville

JeffRockatansky, 2022-08-15:
Having the same issue with ungoogled custom ROM
* LineageOS: 18.1 (2022-08-17)
* Android: 11
* Phone: OnePlus 8 Pro
* GApps: microG core
* Meshtastic app: 1.3.30
* LORA Device: T-Echo
* Meshtastic LORA Firmware: 1.3.30

geeksville, 2022-08-15:
Alas, we use the new google location API.  Removing that dependency is very
low on our priority list.  Though we would welcome pull requests to remove
that dependency.

geeksville, 2022-08-15:
(btw we are totally _keen_ on removing this dependency, it is just almost vanishingly low on our personal interests)

andrekir, 2022-08-15:
except for missing a couple features the app will definitely run on de-googled devices and "instantly crashing" is not common.

andrekir, 2022-08-16:
on Android 8+ Bluetooth results are provided by the OS. some require workarounds like location turned on, some won't work at all unfortunately (like #442). but not an app issue we could fix.

andrekir, 2022-08-16:
> Can an app warn users about this requirement?

there is no requirement. this is a rare OS bug.

geeksville, 2022-08-17:
> there is no requirement. this is a rare OS bug.

I don't think this is quite correct.  For a long time (since kitkat I
think), Android has prevented apps from seeing bluetooth scan results if
the app hasn't been granted location permissions.  This was done to protect
user user privacy because seeing nearby bluetooth macaddrs can be used to
'leak' location info into apps that shouldn't be able to see where the user
is.  I think Kitkat was when facebook began to experiment (now an abandoned
idea) with placing bluetooth 'beacons' in shops etc to let their app know
when a user was in merchant X.

We used to have a dialog that warns the user that they must agree to
location permissions for exactly this reason (but OMG some privacy nuts
thought _we_ were being sleazy because we needed that permission).  I'm not
sure if that dialog is still in there?  Also, we check for that permission
via the google-play service, so possibly we can't know about our missing
location permission on !googlish phones.

andrekir, 2022-08-17:
@geeksville to address those privacy concerns you mention android 8 added [companion device pairing](https://developer.android.com/guide/topics/connectivity/companion-device-pairing), where the OS does the scanning and there is no location requirement. we use it when available. classic scan with dialog still there when not.

geeksville, 2022-08-18:
That's super great I'm glad you added support for that new api!