meshtastic/Meshtastic-Android #178: can not change name or send messages

datadrian, 2020-09-29:
Greetings, I'm running the android app on an acer r11 chrome book, didnt have this issue yesterday but today I do, this issue is occurring in both released version and the beta version.

App connected to heltec 1.0-us/1.1.0

can not change name and can not send messages.
I see some of the incoming messages
I can type in the send message text box, but upon hitting enter no message sends. Worked fine yesterday.

I have submitted a bug report.

Thanks! and great work!!!

geeksville, 2020-10-08:
alas - the chromebook bluetooth stack for android is notoriously buggy.  I'll take a look at this in a couple of weeks after some other work items are finished.

datadrian, 2020-10-08:
Understood, thank you.

Adrian Stucker
513.341.LENS

andrekir, 2022-02-22:
closing. if this issue is still present please reopen.

closed 2022-02-22 by andrekir