meshtastic/Meshtastic-Android #185: Functional requirements: Add notification sounds to the APP

LuckyFishGeek, 2020-10-08:
so you don't have to stare at the screen and wait for messages

DylanHoen, 2020-10-13:
I guess you would want the ability to turn sound or vibration on or off, but maybe that setting is part of the android settings, rather than the app.  The app has a pretty simple interface.  Maybe it needs an Advanced tab with a scroll bar for a bunch of less commonly used settings.

jeremiah-k, 2020-10-29:
> so you don't have to stare at the screen and wait for messages

Notifications are the only major function that is lacking.  If this was implemented, the system would be very usable.

mc-hamster, 2021-11-18:
andrekir — Today at 11:05 AM
great!
https://github.com/meshtastic/Meshtastic-Android/issues/96 (Note sounds OK)
https://github.com/meshtastic/Meshtastic-Android/issues/154 (date stamp OK)
https://github.com/meshtastic/Meshtastic-Android/issues/185 (Note sounds OK)
https://github.com/meshtastic/Meshtastic-Android/issues/194 (elapsed minutes OK)
https://github.com/meshtastic/Meshtastic-Android/issues/258 (Dark theme OK)
https://github.com/meshtastic/Meshtastic-Android/issues/279 (User reports resolved)
https://github.com/meshtastic/Meshtastic-Android/issues/111 (Solved in PR #234)
https://github.com/meshtastic/Meshtastic-Android/issues/281 (Solved on 1.2.45)
https://github.com/meshtastic/Meshtastic-Android/issues/289 (Solved on 1.2.45)

closed 2021-11-18 by mc-hamster