meshtastic/Meshtastic-Android #186: Add stop app button/reboot start option

Bussemann, 2020-10-08:
Once the app starts, it will stay on until the phone is rebooted, then it stays off.

If this is for securing that messages are receieced, it should restart at reboot, so you don't miss any messages. 

Also, a stop button that cleanly exits the app so when things go wonky you don't have to go through android/settings/app/force termination.

geeksville, 2020-10-21:
Hi @Bussemann - thanks for the writeup.

> Once the app starts, it will stay on until the phone is rebooted, then it stays off.

It will keep the service running only as long as a device is selected in the device list (on the settings page).

Activities etc... don't consume any (appreciable) resources on Android because they will be killed by Android if needed and then reinflated if the user goes back to them.

> If this is for securing that messages are receieced, it should restart at reboot, so you don't miss any messages.

It does restart the service (only) at boot (if you have a radio selected).  It does this so that notifications and messaging works.

> Also, a stop button that cleanly exits the app so when things go wonky you don't have to go through android/settings/app/force termination.

I'm not such a fan of this idea because a) if things are wonky it is better for the user to be encouraged to file a bug, b) android apps generally shouldn't have exit buttons.

hestela, 2020-10-22:
would it be possible to stop the bluetooth service so that at least the notification goes away? either through a button or just when you "close" the app by swiping it away. Also when there are some minor issues i have been able to fix them with closing the app through the system settings

geeksville, 2020-11-11:
Hi @hestela - the bluetooth service needs to stay running to connect to devices.  But if you select "None" in the settings tab for the device the service should stop and the notification will go away.  Does that help?

geeksville, 2021-01-03:
ok closing for now - because app seems pretty stable and android doesn't need exit buttons on apps (usually)

closed 2021-01-03 by geeksville