meshtastic/Meshtastic-Android #188: QR code gives 404

stephen304, 2020-10-08:
I have the meshtastic app 1.1.02 from google play and ttgo tbeam with fw 1.1.2, trying to connect with a friend but with the same channel name we get different qr codes and don't see each other. When one of us shares our qr code and the other scans it, it just opens up the github pages 404 page.

@geeksville

stephen304, 2020-10-08:
Well, I uninstalled and reinstalled the app and left the beta program, on 1.1.0 and it works now :upside_down_face:

geeksville, 2020-10-09:
Hmm strange.  I'll play with it here.  Thanks for the report.

justinlyman, 2020-11-14:
I am experiencing this same issue, generated URL results in 404 error.
Android App Version: 1.1.5 (Google Play Store)
Devices: tbeam with firmware version 1.1.8
Here is an example of the generated URL: https://www.meshtastic.org/c/#GAEiID5GRpdzZfwswxtsZSE-yrZeFcJrGLpjjpPFKVUa2CkfKgR0ZXN0

justinlyman, 2020-11-14:
> I am experiencing this same issue, generated URL results in 404 error.
> Android App Version: 1.1.5 (Google Play Store)
> Devices: tbeam with firmware version 1.1.8
> Here is an example of the generated URL: https://www.meshtastic.org/c/#GAEiID5GRpdzZfwswxtsZSE-yrZeFcJrGLpjjpPFKVUa2CkfKgR0ZXN0

Joined the beta, and still experiencing the issue on Version 1.1.8 of the android app.

geeksville, 2020-11-14:
Hmm.  Those urls should cause the Android app to open when you share them
with some other user.  If you click to open them they are not launching the
Android app?

(Sent from a phone - please ignore typos)

dave032285, 2020-11-17:
I don't know if this helps but i was able to get it to open up Via the app formatted like this.. 
https://www.meshtastic.org/c/#GAEiID5GRpdzZfwswxtsZSE-yrZeFcJrGLpjjpPFKVUa2CkfKgR0ZXN0=

geeksville, 2020-11-17:
Ooh that does help!  You just added the = terminal character?

(Sent from a phone - please ignore typos)

dave032285, 2020-11-17:
yes but bad news it doesnt change the channel.. I'm about to set you up with bug reports from a couple phones and I have 6 modules up and running I'm just going to try and change the channels with the barcodes and with the URLs than report I'm running 1.1.8 App and firmware. hop it helps lemme know

dave032285, 2020-11-17:
just for fun I was up all night trying to figure out where all that stuff would be in the app and possibly how to fix it.. fixing its out of the question for now. I barely have it up on the emulator lol. but just out of curiosity could this issue possible persist in  **\Meshtastic-Android\app\src\main\java\com\geeksville\mesh\model\Channel.kt,  its a long shot but thought asking wouldn't hurt lol

geeksville, 2020-11-17:
@dave032285 thanks - I'll try reproing this before.  Your tip was quite helpful!  Can you tell me your android version number?  I'll make an emu instance that matches.

dave032285, 2020-11-17:
that was anti-climatic.. it just worked.. you do it the fancy way,  Ill get setup with 1.1.5 and run 1.1.8 firmware good ole hardware, lol I guess that's where I should have started.. ttyl

dave032285, 2020-11-18:
App 1.1.5 URL worked tried 1.1.5 shared to 1.1.8 and vise versa both opened and changed channels.. even though it worked I still submitted the bug reports. ill mess with it some more tonight.. ill refrain from submitting anything else unless I duplicate the issue

dave032285, 2020-11-18:
@geeksville figured out how to reproduce this every time, here shortly ill edit this post with a write up of how u can duplicate it and how one might accidentally come across this.

geeksville, 2020-11-18:
oh yes - that makes sense!  We do use an API call to say "don't even give the user the choice, always open meshtastic", but that feature wasn't added until recent Androids.  What version (8.1 etc) of android are you running?

geeksville, 2020-11-18:
btw @stephen304, the investigation by @dave032285 should probably help you also.

dave032285, 2020-11-18:
@geeksville for this both devices were running Android 10, that API is effective for sure. There are a few ways to accidentally make it do that it doesn't get called.. Im curious if this works for @justinlyman

justinlyman, 2020-11-20:
> @geeksville for this both devices were running Android 10, that API is effective for sure. There are a few ways to accidentally make it do that it doesn't get called.. Im curious if this works for @justinlyman

I will give it a try tonight.

dave032285, 2020-11-20:
Awesome!

geeksville, 2020-11-21:
good to know - I'll investigate today to see if perhaps we are not setting trusted app correctly.

dave032285, 2020-11-21:
> good to know - I'll investigate today to see if perhaps we are not setting trusted app correctly.

Im sorry,  android:autoVerify="true is correctly implemented it can just be over written by the user I personally do this for the Amazon and eBay. Because I can't stand browsing with the app Amazon and eBay app But the apps are convenient for messages.
I personally didn't do it for meshtastic except to give a example of how one might consistently get the 404 error.

justinlyman, 2020-11-21:
So messed around with it a bit tonight, and figured something out. I checked the app setting and my app was already configured correctly to open the links with the app. What I discovered is if I use a QR scanner app, it launches the app and connects to the channel properly. If I take the same generated url, and paste it into chrome, it just gives me the 404 error. This looks to be an Android bug to me. I am running a cheap phone with Android 10.

justinlyman, 2020-11-21:
> So messed around with it a bit tonight, and figured something out. I checked the app setting and my app was already configured correctly to open the links with the app. What I discovered is if I use [...]

Also tried pasting the link into a text editor and it launches the app correctly from there, this problem seems to be specific to web browsers(I tried firefox as well).

dave032285, 2020-11-21:
Thanks! My theory From what I gathered reading about the APi is it looks like it will only open up in meshtastic if you click a link or scan the qr code unless changed in permission manager. Because the user never specified/doesn't get the option to choose, however if you open the browser and paste the link your overriding that default and telling android to use the browser not meshtastic,  Kind like the picture. 
[image]

justinlyman, 2020-11-23:
> Thanks! I "think" the reason for the 404 error when you copy and paste in a browser is because by manually pasting the URL in the browser your telling android that you want to use the browser to handle the URL. The "API-Don't even give the user the choice" happens automatically before the user would choose the app, but if the user specifies before the API won't get called.
> I could be way off on this or just flat out not making since. Im new to all this and just trying to be as helpful as possible while i learn.. gotta start somewhere right? Lol
> [image]

I think you are spot on, and it is working as intended. I do think it is a bit confusing having the functionality in the app to copy a url when you can't open it directly in a browser, as this was my first thought of what to do was a url. Maybe there should be some kind of disclaimer or something added to the copy link feature letting people know this is the case. Definitely more of an edge case, but something that should probably be addressed eventually.

dave032285, 2020-11-24:
The copy and paste function comes in handy when u want to send the info to a person via a message. For some it might be easier to click the link than figure out how to scan the qr code thats on the device you intend to use.. Not everyone like todo things the "hard" way like we all here clearly enjoy.

geeksville, 2020-11-28:
hmm - in my testing I think the app is doing the trusted app thing correctly.  It defaults to open the link in meshtastic and doesn't ask the user which app to use.  Is it possible you selected to "ask" in the android settings for this?

> An Android App Link is a deep link based on your website URL that has been verified to belong to your website. So clicking one of these immediately opens your app if it's installed—the disambiguation dialog does not appear. Though the user may later change their preference for handling these links.

https://developer.android.com/training/app-links/verify-site-associations

dave032285, 2020-11-28:
> hmm - in my testing I think the app is doing the trusted app thing correctly. It defaults to open the link in meshtastic and doesn't ask the user which app to use. Is it possible you selected to [...]

Me personally yes I did change it in the settings. That was the only way to reproduce the issue.. I personally didn't experience this issue I was just trying to help. Thats kinda what I was getting at. It's would have had of been manually changed in settings or running a older android version..

geeksville, 2020-11-28:
ok thanks!  I'll repurpose your image into an entry in the manual

insanity54, 2021-04-10:
I have one Android 9 phone without GSF/Google Play Store running Meshtastic 1.2.17, and another with Android 10 that does have Google Play Store with Meshtastic 1.2.22. I don't see a way for the Android 9, non-GSF phone to open the meshtastic.org/c/... URL in the Meshtastic app. If both phones had that configuration, I'm not sure it would be possible to join the same channel.

I wish there was the option to paste or just type the channel ID.

andrekir, 2021-12-19:
in-app QR code scanner added in version 1.2.51 plus detailed step-by-step on how to open and share channel links: https://meshtastic.org/docs/software/android/android-usage#join-a-channel

closed 2021-12-19 by andrekir