meshtastic/Meshtastic-Android #194: Add date to time user was last heard from

DylanHoen, 2020-10-13:
In the android app, 2nd tab from the left (I guess you would call it users), there is a timestamp in the lower right corner for each user.  I assume that that is the last time any update was received from that user.  It is just time of day, so I don't know if contact was made today, yesterday, or last week.  It should include the date, too.

DylanHoen, 2020-10-13:
Oh, the time passed could be just as useful as the full datetime:
https://github.com/meshtastic/Meshtastic-Android/issues/191

balazs-h, 2020-10-19:
Yes, I think human readable format like google maps would be much useful. As I see there are several duplicate of my issue. https://github.com/meshtastic/Meshtastic-Android/issues/153 So its clear that this "last_time/accuracy topic" is a key to the reliability of the whole system.

geeksville, 2020-10-19:
good idea - yeah internally the time is represented as # of seconds since 1970.  So it would be quite possible to do a nice human readable version for things not just from today.

mc-hamster, 2021-11-18:
andrekir — Today at 11:05 AM
great!
https://github.com/meshtastic/Meshtastic-Android/issues/96 (Note sounds OK)
https://github.com/meshtastic/Meshtastic-Android/issues/154 (date stamp OK)
https://github.com/meshtastic/Meshtastic-Android/issues/185 (Note sounds OK)
https://github.com/meshtastic/Meshtastic-Android/issues/194 (elapsed minutes OK)
https://github.com/meshtastic/Meshtastic-Android/issues/258 (Dark theme OK)
https://github.com/meshtastic/Meshtastic-Android/issues/279 (User reports resolved)
https://github.com/meshtastic/Meshtastic-Android/issues/111 (Solved in PR #234)
https://github.com/meshtastic/Meshtastic-Android/issues/281 (Solved on 1.2.45)
https://github.com/meshtastic/Meshtastic-Android/issues/289 (Solved on 1.2.45)

closed 2021-11-18 by mc-hamster