meshtastic/Meshtastic-Android #200: latest releases are not published on GitHub

khimaros, 2020-11-03:
as a user of AOSP android, i have no access to the Play store. meshtastic is also not being published to F-Droid.

for these reasons, i am relying completely on the GH releases. the most recent alpha and beta builds have not have not been published to GH which makes it difficult for me to test new releases.

geeksville, 2020-11-11:
ok - sorry about that - I'll push a build later today to github.  I've been kinda lazy about pushing bins there.

Alas - I can't list this project on f-droid because their TOS don't allow analytics (which this project includes as an optional feature).

geeksville, 2020-11-11:
here you go https://github.com/meshtastic/Meshtastic-Android/releases/tag/1.1.8

commit 31ec028f (2020-11-11): Merge pull request #202 from geeksville/dev

1.1.8 generate an APK for github fix #200 (sorry for laziness ;-))
commit 3afbc76c (2020-11-11): 1.1.8 generate an APK for github fix #200 (sorry for laziness ;-))
closed 2020-11-11 by geeksville