meshtastic/Meshtastic-Android #206: Crashing on launch in 1.1.8

Toddhooper, 2020-12-03:
Upgraded two tablets to 1.1.8 app and firmware and the app now crashes within seconds of being opened.

Samsung Active Tab2 running Android 9.0

In the log below, BT is on, mesh device is not present and Wifi is off. Crash occurs regardless of BT, Wifi or device status.

adb logcat | findstr geeksville

12-03 13:44:44.587  5775  5987 I SAPrinter: sendEvent :{cd=det♣1♦packageName♣com.geeksville.mesh, t=ev, en=1001, pn=101, ts=1607031884587}
12-03 13:44:44.597  4632  6062 I ActivityManager: START u0 {act=android.intent.action.MAIN typ=null flg=0x10200000 cmp=ComponentInfo{com.geeksville.mesh/com.geeksville.mesh.MainActivity}} from uid 10056
12-03 13:44:44.605  4632  6062 I ActivityManager: [IOP] sourceDir : /data/app/com.geeksville.mesh-XzCXgCIngfLv26VtzNluEA== in ActivityStacksupervisor
12-03 13:44:44.616  4352  4352 I SurfaceFlinger: id=693 createSurf (2560x2560),2 flag=4, AppWindowToken{52c189b token=Token{376a4aa ActivityRecord{e48ab95 u0 com.geeksville.mesh/.MainActivity t163}}}#0
12-03 13:44:44.626  4632  4632 D GameManagerService: MultiWindowEventListener.onFocusStackChanged(), state=0, top=ComponentInfo{com.geeksville.mesh/com.geeksville.mesh.MainActivity}
12-03 13:44:44.628  4352  4352 I SurfaceFlinger: id=694 createSurf (2560x2560),2 flag=4, 4670250 Splash Screen com.geeksville.mesh#0
12-03 13:44:44.637  4632  4702 V WindowManager: Relayout Window{4670250 u0 Splash Screen com.geeksville.mesh}: viewVisibility=0 req=800x1280 {(0,0)(fillxfill) sim={adjust=pan} ty=APPLICATION_STARTING wanim=0x10302ff
12-03 13:44:44.641  4352  4352 I SurfaceFlinger: id=695 createSurf (800x1280),1 flag=404, Splash Screen com.geeksville.mesh[18132]#0
12-03 13:44:44.643  4632  4702 D WindowManager: makeSurface duration=2 name=Splash Screen com.geeksville.mesh[18132]
12-03 13:44:44.655  4632  6062 D GamePkgDataHelper: notifyAppCreate(), pkgName: com.geeksville.mesh, sendRet: true
12-03 13:44:44.655  4632  4849 D GamePkgDataHelper: getGamePkgData(). com.geeksville.mesh
12-03 13:44:44.655  4632  4849 D GameManagerService: handleMessage(), MSG_APP_CREATE. ignore. pkgName: com.geeksville.mesh
12-03 13:44:44.658  4632  4702 D WindowManager: finishDrawingWindow: Window{4670250 u0 Splash Screen com.geeksville.mesh} mDrawState=DRAW_PENDING
12-03 13:44:44.662  4632  9570 D GamePkgDataHelper: getGamePkgDataIncServer(). com.geeksville.mesh
12-03 13:44:44.662  4632  6062 D GamePkgDataHelper: getSosPolicy(). com.geeksville.mesh
12-03 13:44:44.663  4632  6062 D GameManagerService: identifyGamePackage. com.geeksville.mesh, mCurrentUserId: 0, callerUserId: 0
12-03 13:44:44.663  4632  6062 D GamePkgDataHelper: getGamePkgData(). com.geeksville.mesh
12-03 13:44:44.663  4632  6062 D GameManagerService: noteResumeComponent(), resumeComponent: com.geeksville.mesh
12-03 13:44:44.664  4632  6062 D MARsPolicyManager: onPackageResumedFG pkgName = com.geeksville.mesh, userId = 0
12-03 13:44:44.664  4632  4849 D GameManagerService: handleForegroundChange(). pkgName: com.geeksville.mesh, clsName: com.geeksville.mesh.MainActivity,FgActivityName:com.geeksville.mesh/.MainActivity,userID:0
12-03 13:44:44.664  4632  4849 D GameManagerService:   handleForegroundChange(). set mFgApp: com.geeksville.mesh
12-03 13:44:44.665  4632  4849 D GameManagerService: notifyResumePause(). pkg: com.geeksville.mesh, type: 4, isMinimized: false, isTunableApp: false
12-03 13:44:44.670  4632  6062 I WindowManager: commitFinishDrawingLocked: mDrawState=READY_TO_SHOW Surface(name=Splash Screen com.geeksville.mesh[18132])/@0x941f37c
12-03 13:44:44.670  4632  6062 V WindowManager: Resize reasons for w=Window{4670250 u0 Splash Screen com.geeksville.mesh}:  contentInsetsChanged=false [0,32][0,0] visibleInsetsChanged=false [0,32][0,0] stableInsetsChanged=false [0,32][0,0] outsetsChanged=false [0,0][0,0] surfaceResized=false configChanged=false dragResizingChanged=false reportOrientationChanged=false displayCutoutChanged=false nextForceRelayoutToClient=false
12-03 13:44:44.676  4632  6062 D RemoteAnimationController: createAnimationAdapter(): token=AppWindowToken{52c189b token=Token{376a4aa ActivityRecord{e48ab95 u0 com.geeksville.mesh/.MainActivity t163}}}
12-03 13:44:44.676  4632  6062 I WindowManager: Cancelling animation restarting=true, leash=null, surface=Surface(name=AppWindowToken{52c189b token=Token{376a4aa ActivityRecord{e48ab95 u0 com.geeksville.mesh/.MainActivity t163}}})/@0x5005e8b, parent=Surface(name=Task=163)/@0x878cb68
12-03 13:44:44.676  4632  6062 I WindowManager: Reparenting to leash, surface=Surface(name=AppWindowToken{52c189b token=Token{376a4aa ActivityRecord{e48ab95 u0 com.geeksville.mesh/.MainActivity t163}}})/@0x5005e8b
12-03 13:44:44.677  4352  4352 I SurfaceFlinger: id=697 createSurf (2560x2560),2 flag=4, Surface(name=AppWindowToken{52c189b token=Token{376a4aa ActivityRecord{e48ab95 u0 com.geeksville.mesh/.MainActivity t163}}})/@0x5005e8b - animation-leash#0
12-03 13:44:44.678  4632  6062 D WindowManager: makeSurface duration=2 leash=Surface(name=Surface(name=AppWindowToken{52c189b token=Token{376a4aa ActivityRecord{e48ab95 u0 com.geeksville.mesh/.MainActivity t163}}})/@0x5005e8b - animation-leash)/@0x3d87b81
12-03 13:44:44.678  4632  6062 D SurfaceControl: reparent is called, sc=Surface(name=AppWindowToken{52c189b token=Token{376a4aa ActivityRecord{e48ab95 u0 com.geeksville.mesh/.MainActivity t163}}})/@0x5005e8b, newParentHandle=android.os.BinderProxy@f941426, transaction=android.view.SurfaceControl$Transaction@b945867
12-03 13:44:44.680  4632  6062 V WindowManager: performShowLocked: mDrawState=HAS_DRAWN in Window{4670250 u0 Splash Screen com.geeksville.mesh}
12-03 13:44:44.683  4632  6062 D RemoteAnimationController:     Add token=AppWindowToken{52c189b token=Token{376a4aa ActivityRecord{e48ab95 u0 com.geeksville.mesh/.MainActivity t163}}}
12-03 13:44:44.688  4352  4355 D Layer   : [AppWindowToken{52c189b token=Token{376a4aa ActivityRecord{e48ab95 u0 com.geeksville.mesh/.MainActivity t163}}}#0]::reparent oldParent:[Task=163#0], newParent:[Surface(name=AppWindowToken{52c189b token=Token{376a4aa ActivityRecord{e48ab95 u0 com.geeksville.mesh/.MainActivity t163}}})/@0x5005e8b - animation-leash#0]
12-03 13:44:44.689  4632  6062 I RemoteAnimationController: token=AppWindowToken{52c189b token=Token{376a4aa ActivityRecord{e48ab95 u0 com.geeksville.mesh/.MainActivity t163}}}
12-03 13:44:44.689  4632  6062 I RemoteAnimationController:   leash=Surface(name=Surface(name=AppWindowToken{52c189b token=Token{376a4aa ActivityRecord{e48ab95 u0 com.geeksville.mesh/.MainActivity t163}}})/@0x5005e8b - animation-leash)/@0x3d87b81
12-03 13:44:44.696  4632  9570 D GamePkgDataHelper: getSosPolicy(). com.geeksville.mesh
12-03 13:44:44.713  4632  4649 D CodecSolution: isMaxAspectPackageEx: com.geeksville.mesh : false(0)
12-03 13:44:44.719  4632  4649 D WindowManager: setSystemUiVisibility: vis= 0x18, fullscreenVis= 0x0, dockedVis= 0x0, win=Window{4670250 u0 Splash Screen com.geeksville.mesh}
12-03 13:44:44.722  4352  4352 I SurfaceFlinger:      Client | 0xe7c451c0 | 0002 | RGBA_8888 |   0.0    0.0  800.0 1280.0 |    0    0  800 1280 | Splash Screen com.geeksville.mesh[18132]#0
12-03 13:44:44.724  4632 11393 D GameManagerService: identifyGamePackage. com.geeksville.mesh, mCurrentUserId: 0, callerUserId: 0
12-03 13:44:44.724  4632 11393 D GamePkgDataHelper: getGamePkgData(). com.geeksville.mesh
12-03 13:44:44.742 18132 18132 W geeksville.mes: Accessing hidden method Landroid/content/res/Resources$Theme;->rebase()V (dark greylist, linking)
12-03 13:44:44.743 18132 18132 W geeksville.mes: Accessing hidden method Landroid/content/res/Resources$Theme;->rebase()V (dark greylist, linking)
12-03 13:44:44.844  4632  5046 D MdnieScenarioControlService:  packageName : com.geeksville.mesh    className : com.geeksville.mesh.MainActivity
12-03 13:44:44.852  4352  4352 I SurfaceFlinger:      Client | 0xe7c451c0 | 0002 | RGBA_8888 |   0.0    0.0  800.0 1280.0 |    0    0  800 1280 | Splash Screen com.geeksville.mesh[18132]#0
12-03 13:44:44.901  4632 11393 I WindowManager: reset, Reparenting to original parent, shouldDeferScheduleAnimation=false, destroyLeash=true, surface=Surface(name=AppWindowToken{52c189b token=Token{376a4aa ActivityRecord{e48ab95 u0 com.geeksville.mesh/.MainActivity t163}}})/@0x5005e8b, parent=Surface(name=Task=163)/@0x878cb68, parentNativeHandle=android.os.BinderProxy@c5dc00a, caller=com.android.server.wm.SurfaceAnimator.lambda$getFinishedCallback$0:97 com.android.server.wm.-$$Lambda$SurfaceAnimator$SIBia0mND [... cut ...]
12-03 13:44:50.258  4352  4352 I SurfaceFlinger:      Client | 0xe7c43b40 | 0000 | RGBA_8888 |   0.0    0.0  800.0 1280.0 |    0    0  800 1280 | com.geeksville.mesh/com.geeksville.mesh.MainActivity[18132]#0
12-03 13:44:50.258  4352  4352 I SurfaceFlinger:      Client | 0xe78326c0 | 0000 | RGBA_8888 |   0.0    3.0  485.0  336.0 |  158    0  643  333 | Application Error: com.geeksville.mesh#0
12-03 13:44:50.292  4352  4352 I SurfaceFlinger:      Client | 0xe78326c0 | 0000 | RGBA_8888 |   0.0    3.0  485.0  336.0 |  158    0  643  333 | Application Error: com.geeksville.mesh#0

geeksville, 2020-12-11:
@Toddhooper thanks for the nice log.  Would you be able to capture a version without filtering geeksville?  I suspect it would help provide a clue on the problem (see the four seconds below)

12-03 13:44:45.511 4352 4354 I SurfaceFlinger: id=703 Removed Background for -SurfaceView - com.geeksville.mesh/com.geeksville.mesh.MainActivity@6f32916@0[18132]#0 (0/52)
12-03 13:44:45.513 4632 4643 D WindowManager: finishDrawingWindow: Window{e1e4662 u0 com.geeksville.mesh/com.geeksville.mesh.MainActivity} mDrawState=HAS_DRAWN
12-03 13:44:45.525 4352 4352 I Layer : id=703 onRemoved Background for -SurfaceView - com.geeksville.mesh/com.geeksville.mesh.MainActivity@6f32916@0[18132]#0
12-03 13:44:45.525 4352 4352 I Layer : id=702 onRemoved SurfaceView - com.geeksville.mesh/com.geeksville.mesh.MainActivity@6f32916@0[18132]#0
12-03 13:44:45.529 4352 4352 I SurfaceFlinger: Device | 0xe7c43b40 | 0000 | RGBA_8888 | 0.0 0.0 800.0 1280.0 | 0 0 800 1280 | com.geeksville.mesh/com.geeksville.mesh.MainActivity[18132]#0
12-03 13:44:45.638 18132 18132 I chatty : uid=10166(com.geeksville.mesh) identical 1 line
12-03 13:44:45.640 4858 4871 D BtGatt.ContextMap: add() - appUid: 10166, appName: com.geeksville.mesh
12-03 13:44:45.646 4858 5269 E BtGatt.GattService: [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: com.geeksville.mesh, scannerId: 9, reportDelayMillis=0
12-03 13:44:45.649 18132 18132 E AndroidRuntime: Process: com.geeksville.mesh, PID: 18132
12-03 13:44:45.649 18132 18132 E AndroidRuntime: at com.geeksville.mesh.ui.SettingsFragment$initClassicScan$3.onChanged(SettingsFragment.kt:10)
12-03 13:44:45.649 18132 18132 E AndroidRuntime: at com.geeksville.mesh.ui.SettingsFragment$initClassicScan$3.onChanged(SettingsFragment.kt:1)
12-03 13:44:46.128 4632 4849 D GameManagerService: handleForegroundChange(). pkgName: com.geeksville.mesh, clsName: com.geeksville.mesh.MainActivity,FgActivityName:com.geeksville.mesh/.MainActivity,userID:0

*** interesting bit probably happens in these 4 seconds ***

12-03 13:44:49.657 4632 11393 W ActivityManager: crash : com.geeksville.mesh,0
12-03 13:44:49.660 4632 11393 W ActivityManager: Force finishing activity com.geeksville.mesh/.MainActivity
12-03 13:44:49.695 4632 4649 I ActivityManager: Showing crash dialog for package com.geeksville.mesh u0
12-03 13:44:49.704 5806 18247 D AppErrorNotification: errorType : 24, process : com.geeksville.mesh , uid : 0
12-03 13:44:49.736 4352 4352 I SurfaceFlinger: id=705 createSurf (2560x2560),2 flag=4, d5659 Application Error: com.geeksville.mesh#0
12-03 13:44:49.738 4632 4649 V WindowManager: Changing focus from Window{e1e4662 u0 com.geeksville.mesh/com.geeksville.mesh.MainActivity} to Window{d5659 u0 Application Error: com.geeksville.mesh} Callers=com.android.server.wm.WindowManagerService.addWindow:1810 com.android.server.wm.Session.addToDisplayForTranslate:244 android.view.ViewRootImpl.setView:1014 android.view.WindowManagerGlobal.addView:381

geeksville, 2020-12-11:
hmm - there are a number of mapbox reports of issues with samsung tablets that have been fixed in the last few months.  So I just went ahead and updated to the current version of the mapbox code  and pushed out a 1.1.21 build to the alpha/beta channel.

Toddhooper, 2020-12-11:
Thanks. I updated to 1.1.2.1 but still seeing crash on both. Here's the full logcat.
[logcat.log.log](https://github.com/meshtastic/Meshtastic-Android/files/5679852/logcat.log.log)

Toddhooper, 2020-12-11:
Also noticed this BT crash repeating in an earlier log.

--------- beginning of crash
12-03 13:23:58.939 15934 15934 E AndroidRuntime: FATAL EXCEPTION: main
12-03 13:23:58.939 15934 15934 E AndroidRuntime: Process: com.geeksville.mesh, PID: 15934
12-03 13:23:58.939 15934 15934 E AndroidRuntime: java.lang.IllegalArgumentException: null is not a valid Bluetooth address
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at android.bluetooth.BluetoothDevice.<init>(BluetoothDevice.java:1176)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at android.bluetooth.BluetoothAdapter.getRemoteDevice(BluetoothAdapter.java:985)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at com.geeksville.mesh.ui.SettingsFragment$initClassicScan$3.onChanged(SettingsFragment.kt:10)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at com.geeksville.mesh.ui.SettingsFragment$initClassicScan$3.onChanged(SettingsFragment.kt:1)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.lifecycle.LiveData.considerNotify(LiveData.java:6)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.lifecycle.LiveData.dispatchingValue(LiveData.java:5)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.lifecycle.LiveData$ObserverWrapper.activeStateChanged(LiveData.java:9)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.lifecycle.LiveData$LifecycleBoundObserver.onStateChanged(LiveData.java:5)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.lifecycle.LifecycleRegistry$ObserverWithState.dispatchEvent(LifecycleRegistry.java:3)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.lifecycle.LifecycleRegistry.sync(LifecycleRegistry.java:50)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.lifecycle.LifecycleRegistry.moveToState(LifecycleRegistry.java:5)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.lifecycle.LifecycleRegistry.handleLifecycleEvent(LifecycleRegistry.java:2)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.fragment.app.Fragment.performStart(Fragment.java:10)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.fragment.app.FragmentManager.moveToState(FragmentManager.java:110)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.fragment.app.FragmentManager.moveFragmentToExpectedState(FragmentManager.java:4)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.fragment.app.BackStackRecord.executeOps(BackStackRecord.java:26)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:51)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:10)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.fragment.app.FragmentManager.execSingleAction(FragmentManager.java:5)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.fragment.app.BackStackRecord.commitNow(BackStackRecord.java:3)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.viewpager2.adapter.FragmentStateAdapter.placeFragmentInViewHolder(FragmentStateAdapter.java:22)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.viewpager2.adapter.FragmentStateAdapter.onViewAttachedToWindow(FragmentStateAdapter.java:2)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.viewpager2.adapter.FragmentStateAdapter.onViewAttachedToWindow(FragmentStateAdapter.java:1)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.recyclerview.widget.ChildHelper.addView(ChildHelper.java:12)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.recyclerview.widget.RecyclerView$LayoutManager.addViewInt(RecyclerView.java:23)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.recyclerview.widget.LinearLayoutManager.layoutChunk(LinearLayoutManager.java:6)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.recyclerview.widget.LinearLayoutManager.fill(LinearLayoutManager.java:12)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.recyclerview.widget.LinearLayoutManager.scrollBy(LinearLayoutManager.java:7)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.recyclerview.widget.LinearLayoutManager.scrollHorizontallyBy(LinearLayoutManager.java:2)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.recyclerview.widget.RecyclerView.scrollStep(RecyclerView.java:6)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at androidx.recyclerview.widget.RecyclerView$ViewFlinger.run(RecyclerView.java:24)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:988)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at android.view.Choreographer.doCallbacks(Choreographer.java:765)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at android.view.Choreographer.doFrame(Choreographer.java:697)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:967)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:873)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:214)
12-03 13:23:58.939 15934 15934 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:7179)
[... cut ...]
12-10 23:33:42.046 25506 25506 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:214)
12-10 23:33:42.046 25506 25506 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:7179)
12-10 23:33:42.046 25506 25506 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
12-10 23:33:42.046 25506 25506 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
12-10 23:33:42.046 25506 25506 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:975)

geeksville, 2020-12-12:
Thanks!  That log was super useful.  Though I notice that it seems to be stripping out all of my apps log printing (even logs printed at ERROR level).  Some !google vendors add non standard logcat filtering to prevent logging by RELEASE build apps.  Can you check to see if you have such an option in your developer settings.  Without those log messages it is hard for me to know exactly what is happening.

I looked at the code I waspuzzled how the parameter to getRemoteDevice() could ever be null.  Actually I think the problem is an error in how we update old app settings and the value was not null but "null".  Can you you try going into the android settings for the app and choose to delete all app data?  I bet it will fix this problem.  If it does that confirms my theory and I'll check in a fix for others.

https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/bluetooth/BluetoothDevice.java#951

Toddhooper, 2020-12-12:
Success! Clearing the data worked, both tablets are working now. I had tried reinstalling the app but makes perfect sense if it was invalid data. Thanks!

geeksville, 2020-12-13:
Thanks for the great report.

closed 2020-12-13 by geeksville