meshtastic/Meshtastic-Android #209: Received messages not showing up in messages tab

timgunter, 2020-12-26:
My device is receiving messages, and they show up in the debug panel but they never make into the messages tab in the GUI. I don't know when or why this started. I have switched my android app between being connected to multiple different devices numerous times. I'm going to try deleting all data and re-installing the app.

timgunter, 2020-12-26:
Did the following and still experiencing the issue

1.  Force stop meshtastic
2.  Clear all data(storage and cache)
3.  Uninstalled meshtastic
4.  Rebooted android
5.  Re-installed meshtastic

I noticed that doing this did not clear all of meshtastic's state. My old messages still appear in the messages tab. How do I clear all meshtastic state and do a completely fresh install? It might be useful if clearing storage did this.

timgunter, 2020-12-26:
Is it possible that clearing data or cache has somehow broken the app? I may have done this in the past as well. I can't remember for sure.

timgunter, 2020-12-26:
I connected the app to a second meshtastic device, and am experiencing the same issue with that device as well. Anything else I should try to provide more/better info? Not sure what else to try.

crossan007, 2020-12-26:
@geeksville any chance this is related to the protobuf message "app" refactor?

timgunter, 2020-12-27:
Maybe, but it's probably a combination of factors(such as having jumped between devices several times, and having cleared the app cache which I think I did at some point prior to experiencing this) otherwise, I think others would have reported it.

ievren, 2020-12-29:
Same issue like @timgunter.
- installed fw:
  firmware-tbeam-EU865-1.1.23.bin from 
  https://github.com/meshtastic/Meshtastic-device/releases/tag/1.1.23
- installed android beta app from play store:
  1.1.22 ??? (are they compatible)
  I checked the permissions in Android and they were set to DENIED:
  -> changed to ALLOWED

Reset device and the device will show a message but android app doesent synchronize somehow?

1. In the notification I can see "Connected 1 of 2 online" ???
2. In the app, I open the "People"-Tab and I cant see anyone else
3. In the app, I open the "Channel"-Tab Device Name is "#-B"
3.1  Unlock wil show "Default"
3.2 Lock will change to "Default-G"
4. In tab "Settings"
4.1 Choosed None (disable) 
4.2 selected Mestastic_b -> Connected to radio (tbeam 1.0-EU865/1.1.23)
5. Reset radio-device

timgunter, 2021-01-02:
@ievren  Do you see the messages in the debug panel? They just never seem to make it into the messages tab in the normal GUI.

geeksville, 2021-01-03:
hmm - I'll try to repro this today.  thanks for the report.

geeksville, 2021-01-04:
I found the problem.  I think the publically release (non alpha/beta test google group) version of the app had a problem talking with alpha device builds.  I'm solving this by promoting the current alpha build to production.

@timgunter can you wait a few hours (for the play store to upgrade), update your android app (and if prompted the device code) and try again?

sorry about that

geeksville, 2021-01-04:
or @levren

timgunter, 2021-01-05:
@geeksville  That seems to have resolved the issue. Thanks!

geeksville, 2021-01-06:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/why-bluetooth-and-not-wifi-ap-mode/2083/6

timgunter, 2021-02-06:
I'm experiencing this issue again. I thought it was because I had a few devices with old firmware(pre device app API refactoring). I have since updated those devices and am experiencing this behavior again. Is there any debug information I can produce that would help track this down?

The messages are showing up in the debug panel, but still never make it into the messages tab in the android app gui.

geeksville, 2021-02-07:
hmm!

geeksville, 2021-02-07:
hmm! @timgunter, oh yes.  So to be clear:

* you see the messages listed in the android debug menu, but not on the text message screen inside the android app?

Could you reproduce this, then click report bug.  Then comment in this bug with your phone model and the time you clicked report bug?  That will allow me to get the logs and it should have what we need.

timgunter, 2021-02-07:
I just clicked report bug.

Yes, the messages show up as TEXT_MESSAGE_APP packets in the debug panel, but they never make it to the text message screen inside the android app.

However, just before I clicked report bug, I paired my phone with one of my heltec boards and updated it to 1.1.34 over BLE. After the update succeeded, I attempted to reproduce the issue, and things were working correctly. The device that is sending the message to my phone is a T-Beam running 1.1.34.

When I paired my phone with another T-Beam(also running 1.1.34) that was exhibiting the issue earlier today, it continued to misbehave in this fashion.

So some devices seem to behave like this when paired with my phone, and others don't.

I like the new text message display! It's really cool that when you jump between being paired with different devices, re-displays things so that messages previously sent by the device you are paired with appear on the right side, and messages from all others appear on the left.

geeksville, 2021-02-07:
@timgunter great thanks!  What model of phone do you have?  (so I can find the report)

timgunter, 2021-02-07:
Oh yeah, sorry. It's a "moto g power" running Android 10.

geeksville, 2021-02-07:
oops one more request - can you reboot your phone and then click on the meshtastic app?  Otherwise the google bug uploader might take up to 48 hrs to send in the report.

timgunter, 2021-02-07:
Just rebooted and clicked on the meshtastic app. Hopefully you'll see it soon.

geeksville, 2021-02-07:
thanks yep - got it now.  here it is.  I'll look at it tomorrow morning (taiwan time)
[com.geeksville.mesh_issue_4518e386cd19614e403892a89decaac6_error_session_601D8017030F000176CD95E156FCB49B_DNE_2_v2.log](https://github.com/meshtastic/Meshtastic-Android/files/5938332/com.geeksville.mesh_issue_4518e386cd19614e403892a89decaac6_error_session_601D8017030F000176CD95E156FCB49B_DNE_2_v2.log)

geeksville, 2021-02-07:
oooh!  this is the cause

```

407
10:31:13.079 AM
com.geeksville.mesh.service.MeshService: insert: packet = from: 2988735004 to: 4294967295 decoded {   data {     portnum: TEXT_MESSAGE_APP     payload: "#@azr rx from @amb:yo snr:10.5 hops:1"   } } id: 2623098454 rx_snr: 6.25 rx_time: 1612665073
408
10:31:13.079 AM
com.geeksville.mesh.service.MeshService$Companion: Broadcasting node change NodeInfo(num=-1306806464, user=MeshUser(id=...740, longName=...ris, shortName=...amb), position=Position(lat=...248, lon=...719, alt=...632, time=1612665073, batteryPctLevel=0))
409
10:31:13.081 AM
com.geeksville.mesh.service.MeshService$Companion: Broadcasting node change NodeInfo(num=-1306232292, user=null, position=Position(lat=...0.0, lon=...0.0, alt=...0, time=1612665073, batteryPctLevel=0))
410
10:31:13.082 AM
com.geeksville.mesh.service.MeshService: Ignoring data from -1306232292 because we don't yet know its ID
```

geeksville, 2021-02-07:
root cause: android app was requiring assigned user names for nodes, if one had never been assigned it would drop texts from that node. The android app normally assigns a name to any node it directly connects to, but nodes that have only talked to the python api probably haven't been given a name.  fixing.

timgunter, 2021-02-07:
All my nodes have been assigned names. Is this because there is an issue with the node DB not being updated correctly or something?

timgunter, 2021-02-07:
Yes, this workaround works.

1) pair with the node who's messages are not showing up
2) tweak the node name from the android app
3) pair with the original device again

Messages from the "tweaked" device will now show up in the messages tab.

commit 65e91c46 (2021-02-08): fix #209.  don't ignore messages from nodes that only have partial nodeinfo
closed 2021-02-08 by geeksville