meshtastic/Meshtastic-Android #212: look for autobug from "samsung s7" update failed

geeksville, 2021-01-05:
from @iz1iva

I’m trying to update a LoRa32 and a T-Beam, both on 1.1.31, but as I press the “update to 1.1.32” button in the app, nothing happens. I’ve never used the update facility in the app, perhaps I’m missing something?

at this timestamp: https://meshtastic.discourse.group/t/android-app-new-releases/78/81

geeksville, 2021-01-05:
hmm - it seems like most of the log messages are stripped.  currently googling to see if there is some sort of special samsung stuff they added to prevent most logcat.

[com.geeksville.mesh_issue_4518e386cd19614e403892a89decaac6_error_session_5FF32DAA025F0001358BECF3DA6E2AD6_DNE_1_v2.log](https://github.com/meshtastic/Meshtastic-Android/files/5767876/com.geeksville.mesh_issue_4518e386cd19614e403892a89decaac6_error_session_5FF32DAA025F0001358BECF3DA6E2AD6_DNE_1_v2.log)

geeksville, 2021-01-05:
oh I think I found the remote logging problem.  doing a test now.

geeksville, 2021-01-05:
hmm no - that wasn't it @iz1iva - do you normally run with "analytics disabled" in the app and you turned it on for just this crash report?  I'm thinking there might be some init I need to redo in that case.

IZ1IVA, 2021-01-05:
> hmm no - that wasn't it @IZ1IVA - do you normally run with "analytics disabled" in the app and you turned it on for just this crash report? I'm thinking there might be some init I need to redo in that case.

I've run the app with analytics enabled since a few weeks now.

geeksville, 2021-01-06:
hmm - that's strange.  I'm a bit puzzled.  I did just make logging less conditional, so if/when this shows up again (on 1.1.34 again) I'll hopefully get a better log.

geeksville, 2021-02-06:
ok seems solid now (I think - no new reports in a long time)

closed 2021-02-06 by geeksville