meshtastic/Meshtastic-Android #218: I am getting Could not resolve com.mapbox.mapboxsdk:mapbox-android-sdk:9.6.0.

joeysmart, 2021-01-11:
also adding MAPBOX_DOWNLOADS_TOKEN does not help?
Execution failed for task ':app:compileDebugAidl'.
> Could not resolve all files for configuration ':app:debugCompileClasspath'.
   > Could not resolve com.mapbox.mapboxsdk:mapbox-android-sdk:9.6.0.
     Required by:
         project :app
      > No cached version of com.mapbox.mapboxsdk:mapbox-android-sdk:9.6.0 available for offline mode.
      > No cached version of com.mapbox.mapboxsdk:mapbox-android-sdk:9.6.0 available for offline mode.
      > No cached version of com.mapbox.mapboxsdk:mapbox-android-sdk:9.6.0 available for offline mode.
      > No cached version of com.mapbox.mapboxsdk:mapbox-android-sdk:9.6.0 available for offline mode.

Possible solution:
 - Disable offline mode and rerun the build

jeksys, 2021-02-04:
Having the same issue.

geeksville, 2021-02-06:
alas - mapbox now publishes their jars on their own maven server.  And you need to use your own generated mapbox key to access it.  In the next couple of days I'll update the CI github actions to illustrate how to do this,

geeksville, 2021-02-06:
but a quick workaround for anyone who wants to build now:  Get a downloads token here:

https://docs.mapbox.com/android/maps/guides/install/

Then put that token in your user gradle.properties.

~/development/meshtastic/MeshUtil$ cat ~/.gradle/gradle.properties 
MAPBOX_DOWNLOADS_TOKEN=sk.xxxnZxcXE4dGMwMXVqIn0.P2c0i04lgL0cdCrtRrlQqg

geeksville, 2021-02-06:
relevant issue here: https://github.com/mapbox/mapbox-gl-native-android/issues/630

geeksville, 2021-02-06:
looks like to fix CI I'll need to use: https://docs.github.com/en/actions/reference/encrypted-secrets#creating-encrypted-secrets-for-a-repository

hpsaturn, 2021-02-16:
Thanks @geeksville for you answer in the issue #246 in relation with this. Also I want say that I needed enable in the my Mapbox the `download (read)` option in my Mapbox API configuration section in my account. For other side, I have my secrets like is decribed in the link that you mentioned, if you want I can do a pull request in the workflow with something like that:

```yml
 name: Loading secrets
    ¦ env:
    ¦   mapboxtoken: ${{ secrets.mappoxtoken }}
    ¦   gservices: ${{ secrets.gservices }}
    ¦ run: |
    ¦   echo $gservices > ./app/google-services.json
    ¦   echo $mapboxtoken > ./app/src/main/res/values/mapbox-token.xml
```

geeksville, 2021-02-17:
Omg I would love such a PR!

hpsaturn, 2021-02-19:
it is ready, please review that.

geeksville, 2021-02-20:
Sounds good.  Will do!

(Sent from a phone - please ignore typos)

geeksville, 2021-02-21:
ok thanks @hpsaturn that got us most of the way there.  CI build now fixed!  thanks a zillion (and i got to learn about github secrets)

closed 2021-02-21 by geeksville