meshtastic/Meshtastic-Android #228: No App sync after setting lat lon alt via CLI

drewsed, 2021-02-03:
After setting a fixed position with
meshtastic --setlat 25.2 --setlon -16.8 --setalt 120
text messages aren’t shown anymore in the android app. I can send messages, and they are delivered to all devices, but the phone connected to the device with the fixed position won’t get the message in the app.

geeksville, 2021-02-06:
note to self: (related possible issue - connecting to a node that has fixed position from the android app is causing the assigned lat/lon to be lost?)

geeksville, 2021-02-07:
btw - I think lat/lon setting is not related to the not receiving messages issue (after debugging a bit).  So I'm handling the not rxed messages in this bug instead: https://github.com/meshtastic/Meshtastic-Android/issues/209#issuecomment-774579693

I'll repurpose the current bug

drewsed, 2021-02-07:
Ok, thanks for looking into this, some more info as i played around a bit.

Android 1.1.43 (Cleared data and cache)
Device 1.1.42
(Fresh install 1.1.34 via device-install.sh and update via app to 1.1.42)

Naming both devices, sending some messages, everything works like expected.

Then I set fixed position via cli on one device and after 15 mins the position shows up on the second device/phone.

meshtastic --info shows the correct location...

Messages sent from the fixed position device will be received by the second device/phone, but no ack is received.

Sending messages from second device/phone do not show up in the app (nor debug panel) of the first device, but ack is received by the second one.

after pressing reset button on the first device the fixed position is not shown in meshtastic --info anymore and everything is working again as expected.

geeksville, 2021-02-15:
fixed in the recent 1.1.48 release!  thanks for the report!  (can you confirm it is fixed for you also?)

drewsed, 2021-02-15:
very nice work! Now I can set the fixed position as expected and all nodes show up on the map very quickly. Thanks a lot ;)

closed 2021-02-15 by drewsed