meshtastic/Meshtastic-Android #237: App tab-bar not visible with active keyboard

drewsed, 2021-02-07:
Recently the app tab-bar isn't visible anymore after clicking into the text edit field in the messages tab. 

This behavior shows up on Google pixel 3 (android 11) and xperia xz2 compact (android 10)

[image]

timgunter, 2021-02-08:
I am also experiencing this on a moto g power running android 10. I would prefer if the tab-bar remained visible.

geeksville, 2021-02-09:
interesting!

vfurman-gh, 2021-02-10:
This is probably due to the change in how appearance of soft input is handled. I added adjustPan option here: 
AndroidManifest.xml:
....
android:windowSoftInputMode="stateAlwaysHidden|adjustPan" 
...
The adjustPan helps to bring edit boxes that would be otherwise hidden by the soft input to be visible. I would argue that loosing top bar is not that terrible for gaining visibility into the edit box you are editing....
To reproduce original problem, try editing the lowest edit box on the Settings screen.

drewsed, 2021-02-11:
Ah thanks for that answer, regarding the usability in the settings screen this is indeed a good way to go.

We can now close this issue or leave it open for future reference. Maybe we can add a scrollable panel in the settings tab, because there will be some more settings sometimes in future than inly these two text edits.

drewsed, 2021-02-20:
Ok, after moving the extra text edit boxes to the seperate "Advanced Settings" page it would be nice to revert to the previous mode with the fixed tab bar. what do you think?

I really don't like it to have to close the keyboard every time befor moving to other tabs. So much finger gymnastics, haha

vfurman-gh, 2021-02-22:
Thanks for reminding, here is a PR https://github.com/meshtastic/Meshtastic-Android/pull/250

closed 2021-02-23 by geeksville