meshtastic/Meshtastic-Android #251: Location & Node suggestion

CasualArclamp, 2021-02-26:
Will you be adding a feature to request the location of another node? And to request a node network refresh?

geeksville, 2021-02-26:
Implicitly the oled UI already does this.  If you scroll through to look at
a node it sends a message to that node requesting a fresh position.  But a
UI in the Android app is a good idea.

(Sent from a phone - please ignore typos)

clavo2, 2021-02-26:
Any push/pull query update options, IMO, are a plus for the app.  There can always be an option to enable/disable the function.   It's kinda the point of having an app.   Not being tethered to a 1" screen that may be rooms away, on the roof, etc... Love the last 2 versions.  Most stable I've seen in my test setup so far.  Thanks.

PR #518 (2022-11-06): feat: add option to request remote nodes position 
### Add "Request position" option to nodes list:

[image]

- local node: send a position packet message to remote node with `wantResponse=true`;
- remote node: reply with a position packet in response if it has a valid position.

closes #251 #508
closed 2022-11-06 by andrekir