meshtastic/Meshtastic-Android #263: open / public channel key

hb9eue, 2021-03-22:
Hi Devs

I'm playing around with the 1.2 Version, so what I have observed could be a 'feature' of this Version.

I guess most Meshtastic users want to build 'open' networks everyone could join. I myself have placed a 'headless' device on my roof for that purpose and maybe some could be placed on ham repeater locations. I believe the #default channel always created the same key so any devices using that channel name and same modulation could interconnect.

Now I have noticed, that on my device with 1.2.7, the 'default' channel got renamed to: #LongSlow-V. So I attempted to rename it back to #default, but the Android-App always appends a dash and random letter. Now I can't even get that device back in my existing network as I am not able to rename the channel back to #LongSlow-V (I have not checked out the python script now).

I wonder if that was intended.

-Benoît-

geeksville, 2021-03-22:
hmm - the android and device code adds that suffix letter to serve as a hint for the user that the underlying preshared key is identical (or if the letter does not match not identical/usable).

We do have an ugly hack of if you type "LongSlow" as the channel name it will assume you want to go back to the default channel settings.  But you are correct, we need a clearer UI for that one of us will do that this week (likely by adding a 'trash/reset' icon inside the android UI.

geeksville, 2021-03-22:
also @hb9eue you might dig our forum at: [meshtastic.discourse.group](https://meshtastic.discourse.group).

andrekir, 2021-12-05:
added in 1.2.13

closed 2021-12-05 by andrekir