meshtastic/Meshtastic-Android #269: Voice over LORA.  ...sort of.

DeadlyDad, 2021-03-27:
It just struck me this morning (while pondering another project) that one way to transmit voice over LORA would be to use cellphone apps on either end to do speech to text on one end, send the text, then do text to speech on the other.  (Might it be possible for individuals to set up phoneme databases so that the transmitted voice would be theirs?)  One further advantage would be a transcript of the conversation.  (...and a lot of laughter from autocorrect hijinx.  :grin: )

Does that make sense?

mc-hamster, 2021-03-27:
Moved to the android repository. The request is to use the local device speech to text and text to speech.

andrekir, 2022-10-01:
this now can be done without the need for in-app support.

closed 2022-10-01 by andrekir