meshtastic/Meshtastic-Android #273: Tunnel IP Data over Meshtastic

DakotaNelson, 2021-04-12:
Disclaimer: I know basically nothing about Android development, so this will be an exciting foray. 😛 

The ability to access the mesh [using a TUN driver](https://github.com/meshtastic/Meshtastic-python/issues/35) on the Linux side has me interested in doing similar things on Android. In particular, it would be neat to be able to use something like the Android [VpnService](https://developer.android.com/reference/android/net/VpnService.Builder) to set up a network interface that will route any packets destined for a particular subnet into the mesh.

Any tips/ideas for how to get started on this? I'll be poking around to try to see if I can figure something out but if you have pointers it would likely save me quite a bit of time.

jdstroy, 2021-05-18:
[This](https://github.com/paulmandal/atak-forwarder) is probably a good starting point.  You probably want to look at [IMeshService.aidl](https://github.com/paulmandal/atak-forwarder/blob/main/app/src/main/aidl/com/geeksville/mesh/IMeshService.aidl).

The ATAK-Forwarder integrates Meshtastic through AIDL.  That's one half of the work.

[This](https://github.com/hexene/LocalVPN) might be a reasonable starting point for the other half (namely, attaching to the VpnService functionality).  Take a look at [LocalVPNService.java](https://github.com/hexene/LocalVPN/blob/master/app/src/main/java/xyz/hexene/localvpn/LocalVPNService.java).

There's a good deal more that you likely need to consider beyond the immediate stuff of tunneling IP data: namely, what do you do once you've created an IP link between two Meshtastic radio nodes.  Do you set up a virtual Ethernet-style broadcast network?  Do you set up some IP-level routing?  Do you establish some kind of NAT and default gateway?

jdstroy, 2021-05-19:
A quick glance at Meshtastic-python hints that there's already some implementation of IP tunneling in Meshtastic.  Cool.  It's probably a good idea to make your (Android) implementation compatible with it.

Some things to take a look at:

Looks like you need to configure your app for [IP_TUNNEL_APP as the port and disable ACKs](https://github.com/meshtastic/Meshtastic-python/blob/master/meshtastic/tunnel.py#L201).

Subnet is [configurable](https://github.com/meshtastic/Meshtastic-python/blob/master/meshtastic/tunnel.py#L77) and defaults to [10.115.0.0](https://github.com/meshtastic/Meshtastic-python/blob/master/meshtastic/tunnel.py#L77)/[16](https://github.com/meshtastic/Meshtastic-python/blob/master/meshtastic/tunnel.py#L69).

MTU is set to [200](https://github.com/meshtastic/Meshtastic-python/blob/master/meshtastic/tunnel.py#L102).

[Raw packets](https://github.com/meshtastic/Meshtastic-python/blob/master/meshtastic/tunnel.py#L170) are sent in datagram mode (?) (i.e. Meshtastic is 8-bit clean and requires no encoding; data/payload and bearer/setup channels cleanly separated).

IRainman, 2021-07-29:
It's don't needed because the network is very slow and usable only for text message...

andrekir, 2022-02-22:
keeping settings simple and sticking to defaults for now. closing.

closed 2022-02-22 by andrekir