meshtastic/Meshtastic-Android #277: Nodes info discrepancies between app and device display

IZ1IVA, 2021-04-20:
It's been a while since I first noticed that most nodes are reported in the app as missing position, SNR, last heard or a combination of those. But by checking the device display, I could see all those information, confirming perfect reception. Plus, some of these infos were actually different than those presented in the app.

This makes it harder to debug a mesh, sending messages across, tracking devices, etc. as the mesh seems not to be working while in fact it is!

andrekir, 2021-12-30:
a lot has changed since this was opened, but starting from app [release 1.2.52](https://github.com/meshtastic/Meshtastic-Android/releases) NodeInfo `user`, `position` and `lastHeard` from device and app will match.

andrekir, 2022-01-09:
@IZ1IVA is this issue resolved?

IZ1IVA, 2022-01-10:
> @IZ1IVA is this issue resolved?

I've done limited testing, but apparently it has, thanks very much!

closed 2022-01-10 by andrekir