meshtastic/Meshtastic-Android #279: Meshtastic no longer works on my android

jazap, 2021-05-22:
I downloaded the apk file today since this is no longer available on play store. Meshtastic no longer works for me. Keep getting  meesage: Unexpected bluetooth scan failure: 2

sachaw, 2021-05-23:
Could you post the following information:

Phone model and OS version,
App version,
Hardware used,
Meshtastic firmware version.

geeksville, 2021-05-23:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/meshtastic-app-no-longer-works/3386/2

jazap, 2021-05-23:
> Could you post the following information:
> 
> Phone model and OS version, OnePlus 6, Android 10
> App version,1.2.31
> Hardware used, I have (8) LillyGo T-Beams
> Meshtastic firmware version. 1.2.30
 PROBLEM RESOLVED: After finding some old random posts I turned of Bluetooth, restarted my device, turned Bluetooth back on and Meshtastic worked again.

mc-hamster, 2021-11-18:
andrekir — Today at 11:05 AM
great!
https://github.com/meshtastic/Meshtastic-Android/issues/96 (Note sounds OK)
https://github.com/meshtastic/Meshtastic-Android/issues/154 (date stamp OK)
https://github.com/meshtastic/Meshtastic-Android/issues/185 (Note sounds OK)
https://github.com/meshtastic/Meshtastic-Android/issues/194 (elapsed minutes OK)
https://github.com/meshtastic/Meshtastic-Android/issues/258 (Dark theme OK)
https://github.com/meshtastic/Meshtastic-Android/issues/279 (User reports resolved)
https://github.com/meshtastic/Meshtastic-Android/issues/111 (Solved in PR #234)
https://github.com/meshtastic/Meshtastic-Android/issues/281 (Solved on 1.2.45)
https://github.com/meshtastic/Meshtastic-Android/issues/289 (Solved on 1.2.45)

closed 2021-11-18 by mc-hamster